Koheron/koheron-sdk #29: minor cell improvement

pavel-demin, 2015-12-03:
Hello,

Your `averaging.tcl` and `spectrum.tcl` code looks impressive!

I propose to make this code slightly cleaner by removing all the `list` commands and all the backslashes.

If you replace `foreach {prop_name prop_value} $cell_props` with `foreach {prop_name prop_value} [subst $cell_props]` and `foreach {local_name remote_name} $cell_ports` with `foreach {local_name remote_name} [subst $cell_ports]` in `scripts/project.tcl`, then all the `[list ...]` blocks could be replaced with `{...}` without backslashes.

Best regards,

Pavel

jeanminet, 2015-12-03:
Hello Pavel,

Thanks a lot for the tip ! 
I was starting to get crazy with all this backslashes...

jeanminet, 2015-12-03:
#30

pavel-demin, 2015-12-03:
Correction.

The right command is `[uplevel 1 [list subst $cell_props]]`

jeanminet, 2015-12-03:
It works ! 
Thanks again !

closed 2016-04-14 by jeanminet