Koheron/koheron-sdk #47: Can't build bitstream

tvanderbruggen, 2015-12-25:
When I try to build the bitstream, it fails with:
```
source scripts/bitstream.tcl
# set project_name [lindex $argv 0]
# open_project tmp/$project_name.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/vanderbruggen/ownCloud/zynq-sdk/tmp/cores'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2015.4/data/ip'.
# if {[get_property PROGRESS [get_runs impl_1]] != "100%"} {
#   launch_runs impl_1 -to_step route_design
#   wait_on_run impl_1
# }
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Fri Dec 25 13:43:40 2015...
make: *** [tmp/oscillo.bit] Error 1
```

jeanminet, 2015-12-25:
Il te manque la license. Il faut la choper sur le site de xilinx et ensuite la charger depuis vivado

tvanderbruggen, 2015-12-25:
Oui je m'en suis rendu compte mais j'arrive pas a l'installer.

Je vais regarder. Merci.

closed 2015-12-25 by tvanderbruggen