Koheron/koheron-sdk #86: difficulties with proxies

llombard, 2016-03-02:
During make process, python tries to install automatically yaml from virtualenv, and do not use system wide proxy setting. The installer does not find the server and stops.
Any suggestion?

jeanminet, 2016-03-02:
It seems there is a proxy related issue the version 1.5 of pip:

https://github.com/pypa/pip/issues/1805

You can maybe try to upgrade your version of pip:

`pip install --upgrade pip`

llombard, 2016-03-02:
FIXED. 
Log: pip was already up to date. The way I can install pip packages is:

`sudo pip --proxy http://proxyaddress:80 install jinja2`

but what happens in the make process is:

`virtualenv venv
New python executable in venv/bin/python
Installing setuptools, pip...done.
venv/bin/pip install -r requirements.txt
Downloading/unpacking Jinja2==2.7.3 (from -r requirements.txt (line 1))
  Cannot fetch index base URL https://pypi.python.org/simple/`

of course it cannot fetch the URL because it doesn't use proxy. But by exporting the proxy: 
`export https_proxy=https://proxyaddress:80`
and
`export http_proxy=http://proxyaddress:80`
the makefile eventually worked.

closed 2016-03-03 by jeanminet