Koheron/koheron-sdk #310: PB commandes make et proxy

YoannGoular, 2016-09-13:
Bonjour l'équipe Koheron,

Nous ne parvenons pas à exécuter les commandes make HOST= ... NAME=instrument run car il semble qu'il y ait un problème lors du chargement du bitstream via le serveur sur la carte Red Pitaya. Nous supposons que le proxy local empêche le transfert du bitstream. A noter que la machine de développement et la carte Red Pitaya sont sur le réseau général qui a accès à internet uniquement via un Proxy.

Configuration utilisée :

Nous avons suivi la documentation pour l'installation de la machine virtuelle :

o	image utilisée : led_blinker-37f1e22.img
o	gravée avec : Win32DiskImager-0.9.5-binary

Ces commandes s'exécutent de manière automatique sur la VM pour passer par le proxy de l'office lors du chargement d'une page web :

            export http_proxy=http://proxy.office:80
            <http://proxy.office:80>
            export https_proxy=http://proxy.office:80
            <http://proxy.office:80>
            export ftp_proxy=http://proxy.office:80
            <http://proxy.office:80>
            export no_proxy="localhost,127.0.0.1"

Après avoir vérifié que l'IP de notre Red Pitaya sur le réseau général fonctionnait (avec ping 125.40.20.79), on vérifie que la connection ssh fonctionne  :
--------------------------------------------------------------------------------------------------------
devrp@devrp-VirtualBox:~/koheron-sdk$ ssh root@125.40.20.79
root@koheron:~# 
--------------------------------------------------------------------------------------------------------
Sur un autre terminal 

devrp@devrp-VirtualBox: ~$ cd koheron-sdk/
devrp@devrp-VirtualBox:~/koheron-sdk$. settings.sh 
devrp@devrp-VirtualBox:~/koheron-sdk$ make HOST=125.40.20.79 NAME=adc_dac run
python scripts/make.py --live_zip adc_dac instruments
http://zynq-sdk.s3-website-eu-west-1.amazonaws.com/live-default.zip
[tmp/adc_dac.live] OK
zip --junk-paths tmp/adc_dac-8f7d9c4.zip tmp/adc_dac.bit tmp/adc_dac.live tmp/adc_dac.server.build/kserverd tmp/adc_dac.start.sh tmp/adc_dac.live/*
  adding: adc_dac.bit (deflated 92%)
  adding: kserverd (deflated 61%)
  adding: adc_dac.start.sh (deflated 49%)
  adding: default.js (deflated 69%)
  adding: index.html (deflated 70%)
[tmp/adc_dac-8f7d9c4.zip] OK
curl -v -F adc_dac-8f7d9c4.zip=@tmp/adc_dac-8f7d9c4.zip http://125.40.20.79/api/instruments/upload
*   Trying 125.40.20.79...
* Connected to 125.40.20.79 (125.40.20.79) port 80 (#0)
> POST /api/instruments/upload HTTP/1.1
> Host: 125.40.20.79
> User-Agent: curl/7.47.0
> Accept: */*
> Content-Length: 128353
> Expect: 100-continue
> Content-Type: multipart/form-data; boundary=------------------------da1b3088bb153802
> 
< HTTP/1.1 100 Continue
< HTTP/1.1 504 Gateway Time-out
< Server: nginx/1.10.0 (Ubuntu)
< Date: Thu, 11 Feb 2016 16:35:38 GMT
< Content-Type: text/html
< Content-Length: 192
< Connection: keep-alive
* HTTP error before end of send, stop sending
< 
<html>
<head><title>504 Gateway Time-out</title></head>
<body bgcolor="white">
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.10.0 (Ubuntu)</center>
</body>
</html>
* Closing connection 0
curl http://125.40.20.79/api/instruments/run/adc_dac/8f7d9c4
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.10.0 (Ubuntu)</center>
</body>
</html>

Il semblerait donc que la modification des fichiers de config du serveur n'ait par résolu le problème auquel nous somme confronté, il nous est toujours pas possible d'utiliser les commandes "make ...", au moment d'envoyer le bitstream sur la carte, un blocage s'effectue.

Comment pourrions nous faire en sorte de contourner ce problème ? 

Merci d'avance
Yoann

jeanminet, 2016-09-13:
Merci pour ton retour.

Peux-tu me donner le résultat de ces deux commandes (executées via SSH sur la Red Pitaya):

```bash
journalctl -u uwsgi.service -n 50
```
et
```bash
journalctl -u nginx.service -n 50
```

YoannGoular, 2016-09-13:
voici les résultats:

```
root@koheron:~# journalctl -u uwsgi.service -n 50
-- Logs begin at Thu 2016-02-11 16:28:01 UTC, end at Thu 2016-02-11 16:31:29 UTC. --
Feb 11 16:28:06 koheron systemd[1]: Starting uWSGI...
Feb 11 16:28:06 koheron uwsgi[1998]: [uWSGI] getting INI configuration from /etc/flask-uwsgi/flask-uwsgi.ini
Feb 11 16:29:36 koheron systemd[1]: uwsgi.service: Start operation timed out. Terminating.
Feb 11 16:29:36 koheron systemd[1]: uwsgi.service: Main process exited, code=killed, status=9/KILL
Feb 11 16:29:36 koheron systemd[1]: Failed to start uWSGI.
Feb 11 16:29:36 koheron systemd[1]: uwsgi.service: Unit entered failed state.
Feb 11 16:29:36 koheron systemd[1]: uwsgi.service: Failed with result 'signal'.
Feb 11 16:29:36 koheron systemd[1]: uwsgi.service: Service hold-off time over, scheduling restart.
Feb 11 16:29:36 koheron systemd[1]: Stopped uWSGI.
Feb 11 16:29:36 koheron systemd[1]: Starting uWSGI...
Feb 11 16:29:37 koheron uwsgi[2343]: [uWSGI] getting INI configuration from /etc/flask-uwsgi/flask-uwsgi.ini
Feb 11 16:31:07 koheron systemd[1]: uwsgi.service: Start operation timed out. Terminating.
Feb 11 16:31:07 koheron systemd[1]: uwsgi.service: Main process exited, code=killed, status=9/KILL
Feb 11 16:31:07 koheron systemd[1]: Failed to start uWSGI.
Feb 11 16:31:07 koheron systemd[1]: uwsgi.service: Unit entered failed state.
Feb 11 16:31:07 koheron systemd[1]: uwsgi.service: Failed with result 'signal'.
Feb 11 16:31:07 koheron systemd[1]: uwsgi.service: Service hold-off time over, scheduling restart.
Feb 11 16:31:07 koheron systemd[1]: Stopped uWSGI.
Feb 11 16:31:07 koheron systemd[1]: Starting uWSGI...
Feb 11 16:31:07 koheron uwsgi[2420]: [uWSGI] getting INI configuration from /etc/flask-uwsgi/flask-uwsgi.ini
```

```
root@koheron:~# journalctl -u nginx.service -n 50
-- Logs begin at Thu 2016-02-11 16:28:01 UTC, end at Thu 2016-02-11 16:31:33 UTC. --
Feb 11 16:28:08 koheron systemd[1]: Starting A high performance web server and a reverse proxy server...
Feb 11 16:28:12 koheron nginx[2062]: nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
Feb 11 16:28:12 koheron nginx[2062]: nginx: configuration file /etc/nginx/nginx.conf test is successful
Feb 11 16:28:13 koheron systemd[1]: Started A high performance web server and a reverse proxy server.

```

jeanminet, 2016-09-13:
Merci !
Il semble que `uwsgi` n'arrive même pas à démarrer.
Si tu redémarre ta Red Pitaya, est-ce que tu arrive ensuite à faire un ping les LEDs via l'interface web ?

jeanminet, 2016-09-13:
#311

YoannGoular, 2016-09-13:
En effet le serveur fonctionne car la page html s'affiche.
En revanche la modification du registre config/led n'a pas d'effet.
"ping your board" non plus.

jeanminet, 2016-09-13:
Peux tu essayer ça :
```
$ git pull origin master
$ make HOST=125.40.20.79 app_sync_ssh
```
Puis redémarre ta Red Pitaya

YoannGoular, 2016-09-13:
ok ça marche (reglage led et ping your RP)

jeanminet, 2016-09-13:
Super. Merci pour ton aide !
Tiens-moi au courant si tu arrive à effectuer la commande 
```
$ make NAME=adc_dac HOST=125.40.20.79 run
```

YoannGoular, 2016-09-13:
yep ça marche. merci.

closed 2016-09-13 by jeanminet