Koheron/koheron-sdk #312: RP fonctionne avec DHCP mais pas en local

YoannGoular, 2016-09-13:
en local (sans DHCP), toutes les leds s'allument jusqu'à la 7 inclue et impossible de pinger 192.168.1.255 (ou d'autres). Le PC est bien configuré en 192.168.1.101 et 255.255.255.0. En revanche le DHCP fonctionne.

jeanminet, 2016-09-13:
Si je comprend bien, tu as essayé de connecter la Red Pitaya en direct avec deux méthodes:
* avec une IP statique.
* avec une IP dynamique fournie par un serveur DHCP qui tourne sur ton PC.

C'est bien ça ?

YoannGoular, 2016-09-14:
- DHCP: la RP est connectée et déclarée sur le réseau général. Les leds montrent bien le dernier nombre de l'IP. On peut y accéder sans problème (ping, ssh,...)
- Local: la RP est connectée directement à un port ethernet d'un PC configuré en 192.168.1.101. Les leds sont toutes allumées (255) et impossible de se connecter en SSH ou ping.

jeanminet, 2016-09-14:
Les LEDs affichent 255 par défaut lorsque la carte n'a pas encore trouvé son addresse IP (https://github.com/Koheron/koheron-sdk/blob/master/drivers/common/common.hpp#L83)

Pour se connecter à la Red Pitaya en IP statique, il faut modifier le contenu du fichier `/etc/network/interfaces` sur la Red Pitaya :

## Configuration par défaut (DHCP):
```
allow-hotplug eth0

# DHCP configuration
iface eth0 inet dhcp
  post-up ntpdate -u ntp.u-psud.fr
  post-up systemctl start koheron-server-init
```

## Configuration IP statique (addresse IP = `192.168.1.100`):

```
allow-hotplug eth0

# Static IP
iface eth0 inet static
  address 192.168.1.100
  gateway 192.168.1.0
  netmask 255.255.255.0
  network 192.168.1.0
  broadcast 192.168.1.255
  post-up ntpdate -u ntp.u-psud.fr
  post-up systemctl start koheron-server-init
```

closed 2016-09-17 by jeanminet