Koheron/koheron-sdk #325: command "make NAME=averager INSTRUMENT_PATH=fpga/modules test_module failed

YoannGoular, 2016-09-23:
Bonjour,

J'ai voulu tester le module de moyenne pour comprendre un peu mieux son fonctionnement mais la commande ne semble pas aboutir. J'ai copié la commande que tu indiques sur le quick start pour tester le fonctionnement d'un module :

```
devrp@devrp-VirtualBox:~/koheron-sdk$` make NAME=averager INSTRUMENT_PATH=fpga/modules test_module
Traceback (most recent call last):
  File "scripts/make.py", line 237, in <module>
    dump_if_has_changed(os.path.join('tmp', instrument + '.drivers.yml'), cfg['server'])
KeyError: 'server'
Traceback (most recent call last):
  File "scripts/make.py", line 272, in <module>
    drivers = yaml.load(drivers_file)['server']
KeyError: 'server'
make: ***  Aucune règle pour fabriquer la cible « tmp/averager.config.yml », nécessaire pour « tmp/averager.config.tcl ». Arrêt.
```

jeanminet, 2016-09-23:
Normalement, j'ai corrigé le problème il y a une semaine avec #317. 
Tu peux te mettre à jour en faisant `$ git pull origin master`.

YoannGoular, 2016-09-23:
Super merci ! Désolé pour la redondance du coup

jeanminet, 2016-09-23:
Pas de pb !

closed 2016-09-23 by jeanminet