Koheron/koheron-sdk #339: rp_remote_acquire equivalent?

llombard, 2016-10-14:
Dear Koheron team,

this is more a request than an issue.

I would like to get ADC time series as fast as possible by UDP (the protocol used by rp_remote_acquire that is not compatible with current versions of ecosystem) or, if not possible, by TCP.
UDP and TCP allows to receive data in Labview for example ;-)

example speed test with rp_remote_acquire:
https://github.com/Koheron/laser-development-kit/issues/50

Any solution ?
Thanks :)
Laurent

tvanderbruggen, 2016-10-15:
Dear Laurent,

The limitation with UDP is that packets can be lost, so without any handshaking protocol on top this is not a really reliable way of transferring scientific data. There is probably some situations that would tolerate it and get some speed benefits from getting rid of the handshaking mechanism, but I think this is a small fraction of the applications. It will probably be supported in the future but we don't see any hurry to implement it now.

The current server uses the TCP protocol. so you can access it with any program able to open a TCP socket. The server implements on top of TCP a binary protocol where commands are binary packs of the form:

```
|  0 |  1 |  2 |  3 |  4 |  5 |  6 |  7 |  8 |  9 | 10 | 11 | 12 | 13 | 14 | ...
|      RESERVED     |  class  |  func   |   payload_size    |   payload
```

where the first line indicates the byte position.

Commands are organizes into classes (the driver class) and functions (a function of the driver class) each one of them has an ID send in the command header. The command parameters are packed into the payload.

When a session starts, the client can require the list of the available drivers and their IDs. These data are send in JSON. This way the server defines the contracts with the client and ensure correct data type transmission and provides efficient detection of misuse, avoiding bugs. We use it in our Python client implementation for automatic request generation and error checking.

Note that if you work with a fixed set of drivers then you know in advance the state of the server and don't need to request and parse any JSON data. You can just push the commands in the form described above into the TCP socket.

The commands themselves are a kind of binary version of SCPI commands. We will add some support for string based protocols on the server, and see how we might in the end integrate a full SCPI parser. This will provide flexibility for interfacing with existing code, but will be less efficient than the binary protocol. From the user perspective we will need to define a mechanism to attribute SCPI command names to the driver functions.

llombard, 2016-10-18:
Dear Thomas,

Thanks for your detailed answer. I would probably be very happy with TCP as well.

Would you provide an example of a compatible Labview script or explain how to proceed? :-)

Thanks for your help,
Laurent

jeanminet, 2016-12-21:
Hi Laurent,
The protocol documentation is here: 
https://www.koheron.com/software-development-kit/documentation/koheron-server-protocol

closed 2016-12-21 by jeanminet