Koheron/koheron-sdk #348: Vivado ERROR: [DRC 23-20] Rule violation

YoannGoular, 2016-11-08:
Bonjour l'équipe,

J'utilise actuellement la version d'OS [806de7a](https://github.com/Koheron/koheron-sdk/releases/tag/v0.10.1).
Je cherche à renvoyer des signaux internes vers différentes IO (Extension connectors de la Red Pitaya). J'avais déjà effectué la manip il y a quelques temps et cela fonctionnait très bien.
Je rencontre donc un problème que je n'avais pas avant, sans parvenir à comprendre à quoi cela peut-être dû.

`
ERROR: [DRC 23-20] Rule violation (BIVC-1) Bank IO standard Vcc - Conflicting Vcc voltages in bank 35. For example, the following two ports in this bank have conflicting VCCOs:  
valid_DAC (LVCMOS18, requiring VCCO=1.800) and valid_ADC (LVCMOS33, requiring VCCO=3.300)
`

J'utilise un fichier de contrainte supplémentaire (en plus de ports.xdc et de clocks.xdc) que j'ai créé dans le dossier de mon instrument (ports_trig.xdc) et qui comprend ces quatres lignes : 

```
set_property IOSTANDARD LVCMOS33 [get_ports valid_ADC]
set_property IOSTANDARD LVCMOS33 [get_ports valid_ADC]
set_property PACKAGE_PIN H16 [get_ports valid_DAC]
set_property PACKAGE_PIN H17 [get_ports valid_ADC]
```

J'ai déclaré ce fichier de contrainte dans mon fichier de config.yml de l'instrument : 
```
xdc:
  - boards/red-pitaya/config/ports.xdc
  - boards/red-pitaya/config/clocks.xdc
  - instruments/pulse_generator_moyennage/ports_trig.xdc
```

Je souhaite donc sortir deux signaux différents sur les pins H16 et H17 qui sont alimentées en 3.3V conformément à la [documentation](http://wiki.redpitaya.com/index.php?title=Extension_connectors). 

Je connecte mes "fils" à ces ports de sortie de cette manière : 

```
create_bd_port -dir O valid_DAC
create_bd_port -dir O valid_ADC

connect_bd_net [get_bd_ports valid_DAC] [get_bd_pins averager_delayed/valid]
connect_bd_net [get_bd_ports valid_ADC] [get_bd_pins averager_delayed/first_acq_data]
```
Je pense avoir donc procédé de la bonne manière pour sortir mes signaux mais il semble que l'étape DRC persiste à trouver un conflit dans les tensions d'alimentations de ces pins qui ne sont attribuées que dans mon fichier ports_trig.xdc . Il semble que je ne sois pas le seul à avoir rencontré ce problème [(lien forum)](https://forums.xilinx.com/t5/Implementation/The-Error-DRC-23-20-Rule-violation-BIVC-1-Bank-IO-standard-Vcc/td-p/658471) et qu'il puisse se résoudre en supprimant et rechargeant les fichiers de contraintes et fichiers sources. Malheureusement, en utilisant votre méthode de prototypage, je ne créé pas de projet Vivado au sens propre dans lequel j'ajouterai des fichiers au fur et à mesure et je ne vois donc pas comment je peux procéder pour résoudre cette erreur (?). J'ai tenté de faire un clean_instrument mais cela n'a rien apporté.

Auriez vous des suggestions pour résoudre mon soucis ? 

Merci d'avance

Yoann

jeanminet, 2016-11-08:
Bonjour Yoann,

Dans ton fichier de contraintes il semble que tu as mis deux fois la ligne :
```
set_property IOSTANDARD LVCMOS33 [get_ports valid_ADC]
set_property IOSTANDARD LVCMOS33 [get_ports valid_ADC]
```
au lieu de 
```
set_property IOSTANDARD LVCMOS33 [get_ports valid_DAC]
set_property IOSTANDARD LVCMOS33 [get_ports valid_ADC]
```

Ca pourrait expliquer ton problème.

YoannGoular, 2016-11-08:
Oups .... ! 

Je vais prendre un café, merci beaucoup ^^"

closed 2016-11-09 by minjungkh