Koheron/koheron-sdk #364: Building new fpga cores for the SDK

grahamnaylorccfe, 2016-12-14:
According to the documentation, a new fpga core can be built and tested using a command such as:
$ make CORE=my_core test_core
What are the pre-requisites to be able to run this command? Does it expect a verilog file called my_core in a subdirectory to fpga/cores/test_core along with a core_config.tcl file? When I try this mix I either get an error - "*** No rule to make target...", or if I am in the directory fpga/cores I get the error "Nothing to be done" (and no cores built in the tmp/cores directory). Is something else expected before it will build the core?
thanks
Graham

jeanminet, 2016-12-14:
Hello Graham,

You can follow the structure of the pulse density modulator core in [`fpga/cores/pdm_v1_0`](https://github.com/Koheron/koheron-sdk/tree/master/fpga/cores/pdm_v1_0).
You need:
* a configuration file for Vivado: `core_config.tcl`
* a verilog file for the actual core: `pdm.v`
* a verilog test bench file: `pdm_tb.v` 

You can run the testbench by running the command:
```
$ make CORE=pdm_v1_0 test_core
```

Note the following conventions:
* `_v1_0` suffix to indicate the version of the core (1.0 in our case)
* `_tb` to indicates that the file relates to the testbench

Best,
Jean

grahamnaylorccfe, 2016-12-14:
Thanks Jean,
So I went in to fpga\cores and typed:
make CORE=pdm_v1_0 test_core
thinking this would use the definition in directory pdm_v1_0 (which exists and contains the 3 files) to put a new core called 'test_core' in the tmp\cores directory, however I get the error "No rule to make target 'test_core'. Stop.
I know I am doing something stupid, could you tell me what I should be doing?
thanks
Graham

jeanminet, 2016-12-14:
I'm not sure I understand what you mean.
You need to run the `make CORE=pdm_v1_0 test_core` command from the root of the repository (where the Makefile is).

grahamnaylorccfe, 2016-12-15:
Excellent! Thanks for your patience Jean. It was a simple bit of verilog that I knew worked and so didn't put a test bench file in. Having created a _tb file and included in the directory it works as expected.
Cheers guys
Graham

grahamnaylorccfe, 2016-12-16:
So I was able to generate a target using 'test_core', however I don't seem to be able to apply another name rather than test, and give a version number. I tried things like giving a target of core_v1_0, but that didn't seem to work. What is the command to generate a released version?
thanks
Graham

jeanminet, 2016-12-16:
You can just type something like:
```
$ make tmp/cores/pdm_v1_0
```

grahamnaylorccfe, 2016-12-16:
:+1:   Easy when you know how (durr)! Thanks alot Jean

closed 2016-12-16 by grahamnaylorccfe

jeanminet, 2016-12-16:
I just added a `build_core` target for convenience #365 :
```
make CORE=pdm_v1_0 build_core
```

Note that the cores are build automatically when you build your project as long as they are definfed in the `config.yml` file:
https://github.com/Koheron/koheron-sdk/blob/master/instruments/laser_controller/config.yml#L9