Koheron/koheron-sdk #366: New instrument not starting

grahamnaylorccfe, 2016-12-16:
So I am in business creating new instruments now and so have been able to create the necessary source components for the new instrument called neutron_detector and so ran:
make NAME=neutron_detector
This seemed to proceed correctly, terminating OK with:

................................
Writing bitstream tmp/neutron_detector.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Common 17-83] Releasing license: Implementation
write_bitstream: Time (s): cpu = 00:00:17 ; elapsed = 00:00:16 . Memory (MB): peak = 2149.918 ; gain = 388.734 ; free physical = 4215 ; free virtual = 18637
# close_project
INFO: [Common 17-206] Exiting Vivado at Fri Dec 16 15:33:32 2016...
[tmp/neutron_detector.bit] OK
python scripts/make.py --start_sh neutron_detector instruments
[tmp/neutron_detector.start.sh] OK
python scripts/make.py --live_zip neutron_detector instruments 0.12.0 https://s3.eu-central-1.amazonaws.com/koheron-sdk
[tmp/neutron_detector.live] OK
zip --junk-paths tmp/neutron_detector-58873a7.zip tmp/neutron_detector.bit tmp/neutron_detector.live tmp/neutron_detector.server.build/kserverd tmp/neutron_detector.start.sh tmp/neutron_detector.live/*
  adding: neutron_detector.bit (deflated 87%)
  adding: kserverd (deflated 61%)
  adding: neutron_detector.start.sh (deflated 49%)
  adding: default.js (deflated 72%)
  adding: index.html (deflated 63%)
[tmp/neutron_detector-58873a7.zip] OK

I then uploaded the zip file from the tmp directory using
koheron --host=10.210.1.142 upload neutron_detector-58873a7.zip (with the correct IP for my RP)
The new instrument then appears on the web page of the RP. I click Launch, but the device seems to hang and the View button does not seem to work. A previous simple project to give a count on the LEDs seemed to work. When launching, the new instrument .bit file seems to appear in the /tmp/instrument directory on the RP, however when I try to launch the neutron_detector app the bit file and ...start.sh file seem to still be those for the previously launched instrument. Any idea what is stopping the server and bit file being loaded correctly?
thanks
Graham

jeanminet, 2016-12-18:
Hello Graham,
Did you define the following line in your `config.yml` file:
```yml
server:
  includes:
    - instruments/default/server.yml
    - instruments/default/drivers.yml
```

grahamnaylorccfe, 2016-12-19:
Yes:
I based the config.yml file on the one used for the ADC_DAC (since I intend to use both the ADC and DAC):

---
instrument: neutron_detector
parent: default
board: red-pitaya

cores:
  - axi_cfg_register_v1_0
  - axi_sts_register_v1_0
  - redp_adc_v1_0
  - redp_dac_v1_0

memory:
  - name: config
    offset: '0x60000000'
    range: 4K
  - name: status
    offset: '0x50000000'
    range: 4K

config_registers:
  - led
  - arm_softtrig
  - acquisitionlength
  - simulationpulseamp
  - simulationpulsefreq

status_registers:
  - device_version
  - status
  - state

parameters:
  fclk0: 100000000 # FPGA clock speed in Hz
  dac_width: 14
  adc_width: 14
  n_dac_bram: 2
  n_dac: 2
  n_adc: 2

xdc:
  - boards/red-pitaya/config/ports.xdc
  - boards/red-pitaya/config/clocks.xdc

# Drivers

server:
  includes:
    - instruments/default/server.yml
    - instruments/default/drivers.yml

  drivers:
    - instruments/neutron_detector/neutron_detector.hpp

jeanminet, 2016-12-19:
Your config file looks good.
What happens if you run your instrument using:
```
$ make NAME=neutron_detector HOST=10.210.1.42 run
```

grahamnaylorccfe, 2016-12-19:
When I tried this, I got an error saying that there was no route to the host, however I can upload to the host from the tmp directory, so I followed up by launching it remotely see sequence below:

updating: kserverd (deflated 61%)
updating: neutron_detector.start.sh (deflated 49%)
updating: default.js (deflated 72%)
updating: index.html (deflated 63%)
[tmp/neutron_detector-58873a7.zip] OK
curl -v -F neutron_detector-58873a7.zip=@tmp/neutron_detector-58873a7.zip http://10.210.1.42/api/instruments/upload
*   Trying 10.210.1.42...
* connect to 10.210.1.42 port 80 failed: No route to host
* Failed to connect to 10.210.1.42 port 80: No route to host
* Closing connection 0
curl: (7) Failed to connect to 10.210.1.42 port 80: No route to host
Makefile:155: recipe for target 'run' failed
make: *** [run] Error 7
graham@graham-VirtualBox:~/koheron-sdk$ pwd
/home/graham/koheron-sdk
graham@graham-VirtualBox:~/koheron-sdk$ cd tmp
graham@graham-VirtualBox:~/koheron-sdk/tmp$ koheron --host=10.210.1.142 upload neutron_detector-58873a7.zip
graham@graham-VirtualBox:~/koheron-sdk/tmp$ koheron --host=10.210.1.142 run neutron_detector

The zip file was manually uploaded correctly and appeared in the list of apps in the main web page at http://10.210.1.142, after the run command (last command above) the bit file seemed to be loaded as I had arranged for a pair of LEDs to alternate at 1Hz. When I try and access http://10.210.1.142/live there is however no web page.
Graham

jeanminet, 2016-12-19:
Sorry for not being very helpful.
I don't understand why the `make NAME=neutron_detector HOST=10.210.1.42 run` command does not work while the `koheron --host=10.210.1.142 upload neutron_detector-58873a7.zip` seems to work. Both command use the same HTTP request behind the scene.

What is the content of `neutron_detector-58873a7.zip` ?

grahamnaylorccfe, 2016-12-20:
[neutron_detector-58873a7.zip](https://github.com/Koheron/koheron-sdk/files/663488/neutron_detector-58873a7.zip)
File attached.

jeanminet, 2016-12-20:
Hello Graham,
It looks like the server (`kserverd` in the .zip file) makes Linux crash.
I removed it from the zip file and it does not crash anymore.
My guess is that it tries to access some memory it should not access.
Is there something suspicious in your driver `neutron_detector.hpp` ?

grahamnaylorccfe, 2016-12-20:
/// Neutron Detector driver
///
/// (c) Koheron

#ifndef __DRIVERS_NEUTRON_DETECTOR_HPP__
#define __DRIVERS_NEUTRON_DETECTOR_HPP__

#include <context.hpp>

class NeutronDetector
{
  public:
    NeutronDetector(Context& ctx)
    : cfg(ctx.mm.get<mem::config>())
    , sts(ctx.mm.get<mem::status>())
    {}

   	void set_led(uint32_t led_value) {
        cfg.write<reg::led>(led_value);
    }

    	void set_arm_softtrig(uint32_t arm_softtrig_value) {
        cfg.write<reg::arm_softtrig>(arm_softtrig_value);
    }

	void set_acquisitionlength(uint32_t acquisitionlength_value) {
        cfg.write<reg::acquisitionlength>(acquisitionlength_value);
    }
	
	void set_simulationpulseamp(uint32_t simulationpulseamp_value) {
        cfg.write<reg::simulationpulseamp>(simulationpulseamp_value);
    }
	
	void set_simulationpulsefreq(uint32_t simulationpulsefreq_value) {
        cfg.write<reg::simulationpulsefreq>(simulationpulsefreq_value);
    }
	

	uint32_t get_device_version() {
        return sts.read<reg::device_version>();
    }

    	uint32_t get_status() {
        return sts.read<reg::status>();
    }

	uint32_t get_device_state() {
        return sts.read<reg::state>();
    }

  private:
    Memory<mem::config>& cfg;
    Memory<mem::status>& sts;
};

#endif // __DRIVERS_NEUTRON_DETECTOR_HPP__

grahamnaylorccfe, 2016-12-20:
Didn't get the formatting right so have attached it this time - though had to rename if as a .txt file to be able to uplaod it to this page.
[neutron_detector.txt](https://github.com/Koheron/koheron-sdk/files/664053/neutron_detector.txt)

jeanminet, 2016-12-20:
I don't see anything wrong with the driver.
I would bet that the config and status registers are not properly connected to the AXI interface, which causes a problem when you try to access then with the server.
You may want to check the clocks signals that connect the cfg and status registers in your FPGA design.

grahamnaylorccfe, 2016-12-21:
Thanks for your help - it looks as though the problem was an AXI BRAM in the FPGA design not correctly being interfaced. It works when I remove this. I had actually put an address alocation in the config.yml file (not shown above), but obviously I must have made a mistake somewhere. I will have to study the Oscillo and Spectrum applications a bit more closely. Have a Merry Christmas
Graham

jeanminet, 2016-12-21:
Thanks for the info. 
Merry Christmas to you too.

closed 2016-12-21 by jeanminet