Koheron/koheron-sdk #385: Pulse generator instrument control using test.py

grahamnaylorccfe, 2017-02-10:
In the provided test.py program for the pulse generator instrument, I get the following error:
___________________
~/koheron-sdk$ python instruments/pulse_generator/test.py
Traceback (most recent call last):
  File "instruments/pulse_generator/test.py", line 32, in <module>
    driver.set_dac()
  File "/home/graham/koheron-sdk/instruments/pulse_generator/pulse.py", line 38, in set_dac
    set_dac_data(self, dac_data_1 + 65536 * dac_data_2)
  File "/home/graham/anaconda3/lib/python3.6/site-packages/koheron/koheron.py", line 98, in wrapper
    self.client.send_command(device_id, cmd_id, cmd_args, *args)
  File "/home/graham/anaconda3/lib/python3.6/site-packages/koheron/koheron.py", line 367, in send_command
    cmd = make_command(device_id, cmd_id, cmd_args, *args)
  File "/home/graham/anaconda3/lib/python3.6/site-packages/koheron/koheron.py", line 116, in make_command
    buff += build_payload(args[2], args[3:])
  File "/home/graham/anaconda3/lib/python3.6/site-packages/koheron/koheron.py", line 185, in build_payload
    append_array(payload, args[i], get_std_array_params(arg['type']))
  File "/home/graham/anaconda3/lib/python3.6/site-packages/koheron/koheron.py", line 147, in append_array
    buff += bytearray(array)
TypeError: only integer arrays with one element can be converted to an index
_____________________________________
I am running Python 3.6.0 :: Anaconda 4.3.0 (64-bit) (Ubuntu 16.04 VM).
Graham

grahamnaylorccfe, 2017-02-10:
OK, so I set up an environment to run with python 3.5 and it is good again! (Except that I don't get the graphics as it can't find pygtk and I have to set the backend for matplotlib as 'Agg' which doesn't seem to do anything).

grahamnaylorccfe, 2017-02-10:
Using matplotlib.use('TKAgg') does allow the graphics to show though - hooray!

closed 2017-02-10 by grahamnaylorccfe

tvanderbruggen, 2017-02-10:
I was just trying to get Matplotlib running with Python3. You solved my problem ! Also there is some missing parenthesis in the print.

Anyway, I'm glad it works !

tvanderbruggen, 2017-02-10:
If found similar issues in the decimator test. This should be fix by https://github.com/Koheron/koheron-sdk/pull/386. Thanks for pointing the problem and the solution !

grahamnaylorccfe, 2017-02-20:
Hi Thomas,
Just back from Serre Chevalier.
The upgrade spiral is a nightmare – more like ‘snakes and ladders’!
Hope you didn’t mind me prompting Steve Leibson to mention you in his Xilinx post.
Keep up the good work.
Graham