Koheron/koheron-sdk #387: The server encountered an internal error and was unable to complete your request.

llombard, 2017-02-14:
Dear Koheron developers,

we get the following error:
`devrp@devrp-VirtualBox:~/koheromake NAME=led_blinker HOST=125.40.0.159 run
test -d tmp/koheron_server_venv || (virtualenv tmp/koheron_server_venv && \
			tmp/koheron_server_venv/bin/pip install -r tmp/led_blinker.koheron-server/requirements.txt)
python scripts/make.py --live_zip instruments/led_blinker 0.13.1 https://s3.eu-central-1.amazonaws.com/koheron-sdk
[tmp/led_blinker.live] OK
curl -v -F led_blinker-d2deb66.zip=@tmp/led_blinker-d2deb66.zip http://125.40.0.159/api/instruments/upload
*   Trying 125.40.0.159...
* Connected to 125.40.0.159 (125.40.0.159) port 80 (#0)
> POST /api/instruments/upload HTTP/1.1
> Host: 125.40.0.159
> User-Agent: curl/7.47.0
> Accept: */*
> Content-Length: 103510
> Expect: 100-continue
> Content-Type: multipart/form-data; boundary=------------------------d4564d03d131d166
> 
< HTTP/1.1 100 Continue
< HTTP/1.1 200 OK
< Server: nginx/1.10.0 (Ubuntu)
< Date: Thu, 11 Feb 2016 16:28:46 GMT
< Content-Type: text/html; charset=utf-8
< Content-Length: 44
< Connection: keep-alive
< 
* Connection #0 to host 125.40.0.159 left intact
Instrument led_blinker-d2deb66.zip uploaded.curl http://125.40.0.159/api/instruments/run/led_blinker/d2deb66
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>

`
when trying to follow the procedure on:
https://github.com/Koheron/koheron-sdk
the only difference being Vivado is in 2016.2 instead of 2016.4.

tvanderbruggen, 2017-02-14:
Hi Laurent,

Just to clarify did you build led_blinker with the version 0.13.0 of the SDK and did you use the SD card of the v0.13.0 release ?

Instruments build using older versions of the SDK are not compatible with the new SD card.

llombard, 2017-02-14:
yes, the « koheron-sdk » folder has been « gited » just before compilation and v0.13 OS was running on the RP.

tvanderbruggen, 2017-02-14:
Hi just rebuild led_blinker using Vivado 2016.2 and run it on a freshly burn SD card v0.13.0. It's working fine.

Could you open an SSH shell on your Redpitaya and post the results of the following command lines:
```
journalctl -u uwsgi -n 200
journalctl -u koheron-server -n 200
```

Thank you.

llombard, 2017-02-16:
[j1.txt](https://github.com/Koheron/koheron-sdk/files/780168/j1.txt)
[j2.txt](https://github.com/Koheron/koheron-sdk/files/780169/j2.txt)

Here are the results of the two commands.
Thanks for your help
Laurent

tvanderbruggen, 2017-02-16:
Hi Laurent,

Thanks for the logs.

When I look at it I see that the instruments `decimator` and `oscillo` with version a7b05bb (aka v0.12.0) are running fine but not  `adc_dac` with version d2deb66 (aka v0.13.0).

Given that you build with the v0.13.0 I think your SD card is on version v0.12.0. If you type `<redpitaya_ip>/api/board/version` in your browser what do you get ?

llombard, 2017-02-16:
this was the problem.
now, with 0.13, I get 
`{
  "koheron-sdk": "db9f007"
}
`
and adc_dac compiles successfully

closed 2017-02-16 by llombard