Koheron/koheron-sdk #390: new error since 0.13

llombard, 2017-02-20:
Dear Koheron

Have you seen this kind of error? It prevents successful compilation:
`Generating merged BMM file for the design top 'system_wrapper'...
INFO: [Project 1-111] Unisim Transformation Summary:
  A total of 610 instances were transformed.
  RAM16X1D => RAM32X1D (RAMD32, RAMD32): 4 instances
  RAM32M => RAM32M (RAMD32, RAMD32, RAMD32, RAMD32, RAMD32, RAMD32, RAMS32, RAMS32): 30 instances
  RAM64M => RAM64M (RAMD64E, RAMD64E, RAMD64E, RAMD64E): 448 instances
  RAM64X1D => RAM64X1D (RAMD64E, RAMD64E): 128 instances

open_run: Time (s): cpu = 00:00:22 ; elapsed = 00:00:22 . Memory (MB): peak = 1793.309 ; gain = 765.941 ; free physical = 315 ; free virtual = 8430
# set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
# set_property BITSTREAM.GENERAL.XADCENHANCEDLINEARITY On [current_design]
# write_bitstream -force -file tmp/$instrument_name.bit
Command: write_bitstream -force -file tmp/tof_redpitaya_boite_ll2.bit
Attempting to get a license for feature 'Implementation' and/or device 'xc7z010'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7z010'
Running DRC as a precondition to command write_bitstream
INFO: [DRC 23-27] Running DRC with 4 threads
ERROR: [DRC 23-20] Rule violation (LUTLP-1) Combinatorial Loop Alert - 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: system_i/module_boite_ll2/compteur_moyenne_tvalid_tof/inst/ena_bram_i_1.
WARNING: [DRC 23-20] Rule violation (CHECK-3) Report rule limit reached - REQP-1839 rule limit reached: 20 violations have been found.
`
(... 20 warnings)

`INFO: [Vivado 12-3199] DRC finished with 1 Errors, 21 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
INFO: [Common 17-83] Releasing license: Implementation
5 Infos, 21 Warnings, 0 Critical Warnings and 2 Errors encountered.
write_bitstream failed
ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors.

    while executing
"write_bitstream -force -file tmp/$instrument_name.bit"
    (file "fpga/scripts/bitstream.tcl" line 16)
INFO: [Common 17-206] Exiting Vivado at Mon Feb 20 13:27:38 2017...
Makefile:213: recipe for target 'tmp/tof_redpitaya_boite_ll2.bit' failed
make: *** [tmp/tof_redpitaya_boite_ll2.bit] Error 1
`
This occurs since version 0.13 (used to compile with 0.10).

llombard, 2017-02-20:
The virtal machine is freshly built from :
[https://www.koheron.com/software-development-kit/documentation/setup-development-machine](url)

jeanminet, 2017-02-20:
Hi Laurent, 

Vivado 2016.4 has stricter Design Rule Check (DRC)
The error is :

ERROR: [DRC 23-20] Rule violation (LUTLP-1) Combinatorial Loop Alert - 1 LUT cells form a combinatorial loop. This can create a race condition. Timing analysis may not be accurate. The preferred resolution is to modify the design to remove combinatorial logic loops. If the loop is known and understood, this DRC can be bypassed by acknowledging the condition and setting the following XDC constraint on any net in the loop: 'set_property ALLOW_COMBINATORIAL_LOOPS TRUE [net_nets <myHier/myNet>'. The cells in the loop are: system_i/module_boite_ll2/compteur_moyenne_tvalid_tof/inst/ena_bram_i_1.

In your previous version of Vivado, it was probably only a warning.
* Solution 1 (preferred) : You can try to fix the problem in the `ena` port of the BRAM of your `compteur_moyenne_tvalid_tof` block.
* Solution 2 : Add the line `set_property ALLOW_COMBINATORIAL_LOOPS TRUE` in your xdc file.

llombard, 2017-02-20:
Dear Jean,
- the error appeared also in 2016.2 when we switched the code to the 0.13 version of "koheron-sdk"
- the ena port of the bram is now connected to a constant with the same error
We'll try your solution 2.
Thanks

llombard, 2017-02-20:
Solution 2 works.
Thanks
Laurent

closed 2017-02-21 by minjungkh

mospoly, 2017-07-14:
Dear @jeanminet 

I just tumbled to this post while I was searching for a solution that I have like this in Vivado. I have exactly the same problem of combinatorial loop in my design targeting design of a ring osillator. Can you please let me know in the following command that you have recommended in Solution 2, what should I add to make it run?? [get_nets] ?? [get_pins] ?? [get_ports]  ?? or etc. ??
Thank you so much,

Solution 2 : Add the line set_property ALLOW_COMBINATORIAL_LOOPS TRUE in your xdc file.

jeanminet, 2017-07-18:
Dear @mospoly,
Sorry for the delay.
I don't know the exact command that @llombard used to make it work.