Koheron/koheron-sdk #391: long compilation time

llombard, 2017-02-22:
Short question: since 0.13, compilation time is much longer. Maybe because:

`Phase 4 Post Placement Optimization and Clean-Up

Phase 4.1 Post Commit Optimization
INFO: [Place 46-20] Placer is running with the ExtraNetDelay_high directive. **Post Placement Optimization may take longer to complete with ExtraNetDelay_high compared to other directives**.
INFO: [Timing 38-35] Done setting XDC timing constraints.

Phase 4.1.1 Post Placement Optimization
INFO: [Place 30-746] Post Placement Timing Summary WNS=0.170. For the most accurate timing information please run report_timing.
Phase 4.1.1 Post Placement Optimization | Checksum: 2246f81ac

Time (s): cpu = 00:14:08 ; elapsed = 00:14:12 . Memory (MB): peak = 1982.391 ; gain = 3.012 ; free physical = 3477 ; free virtual = 8666
Phase 4.1 Post Commit Optimization | Checksum: 2246f81ac

Time (s): cpu = 00:14:08 ; elapsed = 00:14:12 . Memory (MB): peak = 1982.391 ; gain = 3.012 ; free physical = 3477 ; free virtual = 8666

Phase 4.2 Post Placement Cleanup
`
Is this directive useful?

jeanminet, 2017-02-22:
Thanks for the feedback !
The directive is useful if you want to be confident with timing issues.
You can now compile quicker in development mode #393 but you need to be careful with timing:

```
make NAME=led_blinker MODE=development
```

jeanminet, 2017-06-09:
The directive `ExtraNetDelay_high` is now removed by default (`MODE=development` by default).
It is only enabled in production mode:

```
make NAME=led-blinker MODE=production
```

closed 2017-06-09 by jeanminet