Koheron/koheron-sdk #397: Koheron upgrade

grahamnaylorccfe, 2017-04-25:
I have been trying to upgrade to version 0.14 so ran:
pip install --upgrade koheron
(I am using python 2.7). The upgrade seems to go OK, but when I type:
koheron version
I get:
 ImportError: No module named python.cli
I can't find any info on what python.cli is.
Any ideas what the issue is?
thanks
Graham

jeanminet, 2017-04-25:
Hello Graham,
Thanks a lot for noticing the problem.
It should be fixed now. c4ca4f3

grahamnaylorccfe, 2017-04-25:
I am just putting together a training video (for my future benefit and new PhD students etc..) and so thought I better start with the latest version. I do see however that a lot of the commands have changed NAME => CONFIG, bd> block_design and the directory structure instruments>examples. No big deal, but I think I will need to re-record a load of sections now sigh…..  No, but not I’m complaining – no, no, no – this is all going to save me a lot of time in the long run!
Anyway, thanks again for all your work!
Graham

jeanminet, 2017-04-25:
Excuse us for the backward incompatibility :(
We had no choice but to break things in order to make the SDK simpler and more maintanable in the long run.
Thank you for your support !
Jean

grahamnaylorccfe, 2017-04-25:
OK, pulled the updated version and run 'make setup' again, however I still get an error when running 'koheron version':

> koheron version
Traceback (most recent call last):
  File "/home/graham/anaconda3/envs/py27/bin/koheron", line 7, in <module>
    from python.cli import cli
ImportError: No module named python.cli

Perhaps I should be using a different command to get the version?

grahamnaylorccfe, 2017-04-25:
OK, so I open a new terminal without anaconda activated. Default python is now 3.6, I get a response from koheron version of 0.13.0
Something is out of sync somewhere.....
I try and run 'make setup' again with this version of python, but get lots of errors. Should I be using python 3.5?
thanks
Graham

grahamnaylorccfe, 2017-04-25:
Durrr, should have done 'sudo make setup' - will try again (maybe even with 'sudo -H make setup')...
G

grahamnaylorccfe, 2017-04-25:
OK, so looking at the warnings when running setup I see that it was expecting to find koheron-sdk/package.json which doesn't seem to be on the github repository.
koheron-sdk/python/version.py does indeed indicate 0.14, but running 'koheron version' uses the koheron file in the anaconda distribution not the default python directory one in /usr/local/bin and so picks up an old version number from somewhere. This is not a problem in itself, but may mean that other things are misdirected now.

jeanminet, 2017-04-25:
Dear Graham,
The problems should be fixed now (171ade9).
`koheron version` should return 0.14.3 and it should work for 2.7 and 3.5.
Sorry for your time.
Jean

grahamnaylorccfe, 2017-04-25:
I removed anaconda and returned to an in-built python version of 2.7. I deleted koheron-sdk and re-cloned a fresh from the current version and after setting up (still with warning about not finding package.json) and chown tmp back to me, I now indeed get a version returned of 0.14.3.

When I try and make the led-blinker instrument using 'make CONFIG=examples/led-blinker/config.yml' (after redirecting fpga/settings.sh to my 2016.2 installation of vivado) I get "recipe for target 'tmp/led-blinker/cores/axi_ctl_register_v1_0/component.xml' failed"
This file is indeed absent from the tmp/led-blinker/cores directory. Do I really need to use Vivado 2016.4?
thanks for your patience.
Graham

jeanminet, 2017-04-25:
From 77f99a3, you can try to run:
`make VIVADO_VERSION=2016.2 CONFIG=examples/led-blinker/config.yml`

We currently use 2016.4 and I checked that it works with 2016.3 but I don't have 2016.2 installed on my machine.

grahamnaylorccfe, 2017-04-26:
I installed 2016.4 and pointed fpga/sttings.sh to that installation and the example works now with no problems. I will try the above trick on another Ubuntu VM machine that has 2016.2 to check for compatibility and will let you know.
Thanks for sorting me out.
Graham

grahamnaylorccfe, 2017-04-26:
Tried on VM on remote server using an installation of Vivado 2016.2 with fpga/settings.sh pointing to this installation and using:
make VIVADO_VERSION=2016.2 CONFIG=examples/led-blinker/config.yml
-Worked fine (just one hiccup with timout connecting to fonts.gstatic.com! but worked on second attempt)
All good.
Graham