Koheron/koheron-sdk #401: Errors while executing ./fpga/vivado/project.tcl

grahamnaylorccfe, 2017-04-28:
Just trying to get my head around all the changes in the latest version. I hope that you expect to maintain the current conventions going forwards?
The changes seem quite profound and porting an old design seems to be taking quite a while (not got to the web part yet...)!

jeanminet, 2017-04-28:
Hello Graham,
What is the problem you are experiencing ?

grahamnaylorccfe, 2017-05-03:
I don't think there is a fundamental issue, I am just going through the pain of upgrading the source files to be compatible with the new standards (cfg -> ctl etc.). Since I was working with block designs exported from Vivado as tcl scripts, it was not as simple, so I had to go back and re-created the block diagrams via your examples. I now need to learn some type script to get a web front-end. Ultimately I will be limited by the network transfer speeds for large acquisition data sets (>200Mbps sustained for several seconds) and so will need to build a local server to buffer the data to RAM disk, or else partition the DDR to send directly to RAM from the PL and map the space in Linux to readout later to the network.
Graham

closed 2017-05-03 by grahamnaylorccfe