Koheron/koheron-sdk #403: Writing modules in vhdl

CHV123, 2017-05-25:
Hi, I was wondering how I would go about writing my modules in vhdl as opposed to verilog. When I tried it I got:

make: *** No rule to make target 'fpga/cores/one_clock_pulse_v1_0/*.v', needed by 'test_core'. Stop.

I tried to find where the verilog specification was in the Makefile but couldn't find it. Any help?

Thanks,
Charlie

jeanminet, 2017-05-25:
Hi Charlie,

The Makefile related to FPGA stuff is [`fpga.mk`](https://github.com/Koheron/koheron-sdk/blob/master/fpga/fpga.mk). It is included from the main Makefile.
I just added support for VHDL cores with #404.

Jean

CHV123, 2017-05-26:
Thanks Jean,

I've implemented the changes and it now works for me.

Charlie

closed 2017-05-26 by CHV123