Koheron/koheron-sdk #407: issue with connect when instrument already loaded

llombard, 2017-06-16:
the command connect('192.168.1.1',name='laser-controller') raises an error if the instrument is already loaded.
same behavior with other instruments.
last version of koheron os, koheron python library and koheron git

```
connect('192.168.1.1',name='laser-controller')
<koheron.koheron.KoheronClient object at 0x7f686ea27b00>
connect('192.168.1.1',name='pulse-generator')
<koheron.koheron.KoheronClient object at 0x7f6879bdaf60>
connect('192.168.1.1',name='pulse-generator')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/osboxes/py3/lib/python3.5/site-packages/koheron/koheron.py", line 53, in connect
    run_instrument(host, *args, **kwargs)
  File "/home/osboxes/py3/lib/python3.5/site-packages/koheron/koheron.py", line 47, in run_instrument
    raise ValueError('Did not found instrument {}'.format(name))
ValueError: Did not found instrument pulse-generator
connect('192.168.1.1',name='laser-controller')
<koheron.koheron.KoheronClient object at 0x7f6879bd5630>
connect('192.168.1.1',name='pulse-generator')
<koheron.koheron.KoheronClient object at 0x7f6879bdaeb8>
connect('192.168.1.1',name='pulse-generator')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/osboxes/py3/lib/python3.5/site-packages/koheron/koheron.py", line 53, in connect
    run_instrument(host, *args, **kwargs)
  File "/home/osboxes/py3/lib/python3.5/site-packages/koheron/koheron.py", line 47, in run_instrument
    raise ValueError('Did not found instrument {}'.format(name))
ValueError: Did not found instrument pulse-generator

```

llombard, 2017-06-21:
````
host = os.getenv('HOST', '192.168.1.1')
client = connect(host, name='led-blinker')
client = connect(host, name='pulse-generator')
````
succeeds when run twice.

```
host = os.getenv('HOST', '192.168.1.1')
client = connect(host, name='pulse-generator')
```

fails when run twice

jeanminet, 2017-06-21:
Thanks.
The problem seems to happen only with python3.

jeanminet, 2017-06-21:
Fixed with 0cb9a07

closed 2017-06-21 by jeanminet