Koheron/koheron-sdk #412: Running cluster instrument

JohnnyMalaria, 2017-06-29:
Hi,

Thank you so much for making your SDK available!

I wish to test the cluster instrument in order to generate synchronized RF signals from two daisy-chained RPs. Both have the Koheron SD card. I have setup a VirtualBox with Ubuntu on my Windows PC and installed all the prerequisites. I can successfully build the instrument and end up with cluster.zip. I assume this is equivalent to the one already on the SD card.

My primary issue is that I don't know how to run it and/or change the line in the Python script for testing (the IP addresses etc for each RP):

`hosts = ['192.168.1.14', '192.168.1.5', '192.168.1.13', '192.168.1.6']`

I'd like to be able to enable/disable using the SATA clock via a bash or python script.

Can you advise me?

Thanks,

John.

jeanminet, 2017-06-29:
Hello John,

The `cluster.zip` you built should indeed be equivalent to the one already on the SD card.

The simplest way of installing the instrument on the Red Pitaya is to run the command 
`$ make CONFIG=examples/cluster/config.yml HOST=192.168.1.42 run`, where HOST is your RP IP address.

You can also use Python. For instance:
```python
from koheron import upload_instrument
hosts = ['192.168.1.1', '192.168.1.2', '192.168.1.3']
for host in hosts:
    upload_instrument(host, 'tmp/cluster/cluster.zip')
```

> My primary issue is that I don't know how to run it and/or change the line in the Python script for testing

You can run the script with `$ python examples/cluster/test.py`.

> I'd like to be able to enable/disable using the SATA clock via a bash or python script.

Could you explain a little more ?

Jean

JohnnyMalaria, 2017-06-29:
Hi Jean,

Thanks for your rapid reply.

Here's an overview of my project that I hope to realize with two RPs:

1. Each RP needs to drive an acoustic-optic modulator at ~40MHz with a precise frequency difference between the two (e.g., 40.000000 and 40.001000MHz). I can do this with one RP and the frequency difference is beautifully stable but I get ~10% intermodulation distortion which severely affects the overall apparatus.

2. I need to generate a combined low-frequency square/sine waveform.

3. All three signals need to be synchronized to a common clock. I want to try putting the two RF signals on separate RPs to see if the intermodulation goes away.

4. One high-speed RP input will be used for capturing a photodetector signal with a frequency centered about the frequency difference between the two RF signals (e.g., 1kHz).

So, I need the cluster functionality, sinewave and arbitrary waveform generation, and ADC at the same time. I assume only one instrument can be loaded into the tmp area at a time(?)

For now it will be sufficient to configure all these functions via a script etc. Eventually, I want to develop either a web instrument or install Jupyter on the RP (with the Koheron image).

The experiment involves a lot of real-time calculations that will be done either on-board the RP or by a Windows app communicating with the RP with your socket server.

It's a bit daunting :) but my first need is to sync the two RPs and see if I can reduce the intermodulation issue. Sadly, if I can't then I will have to try something else instead of RPs.

John.

jeanminet, 2017-06-29:
That looks like an interesting project !

Note that the output buffer of the Red Pitaya ([LT6200](http://cds.linear.com/docs/en/datasheet/62001ff.pdf)) does not have good distortion performance above a few MHz: 

[image]

You may be able to reduce the intermodulation issue with two RP though.

> I assume only one instrument can be loaded into the tmp area at a time(?)

I'm not sure I understand. You can only run one FPGA bitstream at a time.
If you need the functionalities of two instruments, you should develop one that includes both functionalities.

JohnnyMalaria, 2017-06-29:
Thanks, Jean.

I'm not too worried at this time about the distortion. The outputs are fixed single frequency and go through narrow band RF amps that drive the modulators that are also narrow band. The signals look good on my oscilloscope when only one of the two signals is being generated. When both are on, there is amplitude modulation on RF OUT 2 equal to the frequency difference. RF OUT 1 is flat.

What I meant by loading more than one instrument is when the bitstream for the cluster is loaded, can I still access the RP's function generation and high speed ADC. I assume I can since your cluster article shows all the RPs generating an RF signal.

John.

jeanminet, 2017-06-29:
I would suspect that the intermodulation you observe is not an analog issue.
How do you generate the modulations of the FPGA ?

I made a simple instrument that allows to set the frequencies of each RF output (see #413)

[image]

I did not observe the behavior you describe.
Could you try it on your setup ?
[dual-dds.zip](https://github.com/Koheron/koheron-sdk/files/1113195/dual-dds.zip)

Jean

JohnnyMalaria, 2017-06-29:
Hi Jean,

Thanks for testing.

I generate the two RF signals using the Red Pitaya's standard oscilloscope/function generator web app.

I will try the dual DDS. Can you confirm how to install it? Do I just copy the zip to the same location as all the others on the SD card?

John.

jeanminet, 2017-06-29:
> Can you confirm how to install it? Do I just copy the zip to the same location as all the others on the SD card?

Yes, that should work

JohnnyMalaria, 2017-06-29:
Strange.

On a v1.1 RP, after copying the zip and rebooting the app shows up but the links to edit the frequencies (i.e., the blue zeroes) don't do anything. This is on Chrome v59, IE11 and Edge.

On a v1.0 RP, the instrument doesn't show up even though the zip is in the same location and both RPs are using the same SD image.

Anyway, after a lot of troubleshooting, I have discovered that one of my RF amps has an intermittent fault and I suspect the intermodulation is a sign of one of the components being bad. The amps are made by the same company that makes the modulators (in fact they are intended for them) so I maybe able to get a matching replacement amp from them.

So that's one less thing to figure out with the RPs but I still need to sync them so that my 3rd output is sync'd.

John.

jeanminet, 2017-06-30:
Did you copied the zip to `/usr/local/instruments/` ?

JohnnyMalaria, 2017-06-30:
Hi Jean,

I did. I've since put fresh images (0.14.0) on each and now the dual DDS instrument works.

My next step was to try the cluster. I ran the make command you specified from the Ubuntu VirtualBox and I modified test.py to match my IP addresses. I get a Connection Refused error. I assume I have to pass my RP username and password but I don't know how.

```
Traceback (most recent call last):
  File "test_jfm.py", line 56, in <module>
    clients.append(connect(host, 'cluster', restart=False))
  File "/usr/local/lib/python2.7/dist-packages/koheron/koheron.py", line 54, in connect
    client = KoheronClient(host)
  File "/usr/local/lib/python2.7/dist-packages/koheron/koheron.py", line 244, in __init__
    raise ConnectionError('Failed to connect to {}:{} : {}'.format(host, port, e))
requests.exceptions.ConnectionError: Failed to connect to 192.168.2.16:36000 : [Errno 111] Connection refused
```
I modified test.py thus:

```
f __name__=="__main__":
    # Define the IP addresses of the 2 Red Pitayas
    hosts = ['192.168.2.16', '192.168.2.24']
    clients = []
    drivers = []

    for i, host in enumerate(hosts):
        clients.append(connect(host, 'cluster', restart=False))
        drivers.append(Cluster(clients[i]))

    for driver in drivers:
        driver.set_freq(10e6)
        driver.set_clk_source('crystal')
        driver.ctl_sata(1, 0)
        driver.set_pulse_generator(100, 200)

    for i in [1]:
        drivers[i].set_clk_source('sata')

    drivers[0].ctl_sata(1, 0)
    drivers[1].ctl_sata(0, 7)
  
    for i in range(10000):
        drivers[1].phase_shift(1)
        print(i)
        time.sleep(0.01)

```

jeanminet, 2017-06-30:
You don't need to pass username and password.
Could you provide the ouput of the command `$ koheron version` ?

Maybe it is a problem with your Virtualbox network setup. 
You may try running the Python script on Windows.

JohnnyMalaria, 2017-06-30:
Hi Jean,

The version is 0.14.3 (from Ubuntu terminal prompt in VirtualBox). I tried to get 0.15 from git but I ended up at a 404 page.

I can successfully ssh with the RP from within Ubuntu so I assume the network is configured okay.

To run the script under Windows, do I need to copy the SDK somewhere on my Windows OS and then execute the script (I already have Python installed)? Do I have to make anything first?

I do appreciate your help and patience :)

John.

jeanminet, 2017-06-30:
You don't need to copy th SDK to execute the Python script on Windows.
You just to install the koheron package (`pip install koheron` on Windows command line).

JohnnyMalaria, 2017-06-30:
I successfully installed the koheron package and numpy on Windows. I get the same access denied error. Does it matter if I am using Python 2.7 or 3.6?

JohnnyMalaria, 2017-07-01:
Hi Jean,

I ran some more tests. I installed the Koheron package on the RP itself and tried using 127.0.0.1 as the host address in test.py. I ran this using Python on the RP. I got the same connection error.

John.