Koheron/koheron-sdk #415: pulse_width = 8192 in pulse-generator\test.py no not work

llombard, 2017-07-07:
examples\pulse-generator do not work anymore if:
pulse_width = 8192
in test.py (nothing is generated).

llombard, 2017-07-07:
and also, it is not possible to get more than 1024 points from the 8192 pts FIFO at the same time

llombard, 2017-07-07:
also reset_fifo is declared in pulse.py but not in pulse.hpp

jeanminet, 2017-07-07:
I restored `get_next_pulse()` and `reset_fifo()` with cf58c1468.

llombard, 2017-07-10:
Thanks!
however, the compilation now stops (last version koheron-sdk on vivado 2017.1):

```
make --jobs=4 tmp/pulse-generator/serverd 

make[1] : on entre dans le répertoire « /home/devrp/partageLabo/rw/devrp/koh0.15/koheron-sdk »

make[1]: « tmp/pulse-generator/serverd » est à jour.

make[1] : on quitte le répertoire « /home/devrp/partageLabo/rw/devrp/koh0.15/koheron-sdk »

mkdir -p tmp/pulse-generator/web

tsc --pretty --target ES3 --lib es6,dom --alwaysStrict web/koheron.ts web/jquery.flot.d.ts examples/pulse-generator/./web/pulse_generator.ts examples/pulse-generator/./web/app.ts examples/pulse-generator/./web/control.ts examples/pulse-generator/./web/plot.ts --outFile tmp/pulse-generator/web/app.js

[100;30m22[0m let app = new App(window, document, location.hostname, $('#plot-placeholder'));

[100;30m  [0m [91m                                                       ~[0m

examples/pulse-generator/web/app.ts(22,56): [91merror[0m TS2304: Cannot find name '$'.

[100;30m92[0m         this.plot_placeholder.bind('plotselected', (event: JQueryEventObject, ranges: jquery.flot.ranges) => {

[100;30m  [0m [91m                                                           ~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(92,60): [91merror[0m TS2304: Cannot find name 'JQueryEventObject'.

[100;30m114[0m         this.plot_placeholder.bind('dblclick', (evt: JQueryEventObject) => {

[100;30m   [0m [91m                                                     ~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(114,54): [91merror[0m TS2304: Cannot find name 'JQueryEventObject'.

[100;30m124[0m         this.plot_placeholder.bind('wheel', (evt: JQueryEventObject) => {

[100;30m   [0m [91m                                                  ~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(124,51): [91merror[0m TS2304: Cannot find name 'JQueryEventObject'.

[100;30m125[0m             let delta: number = (<JQueryMousewheel.JQueryMousewheelEventObject>evt.originalEvent).deltaX

[100;30m   [0m [91m                                  ~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(125,35): [91merror[0m TS2503: Cannot find namespace 'JQueryMousewheel'.

[100;30m126[0m                                 + (<JQueryMousewheel.JQueryMousewheelEventObject>evt.originalEvent).deltaY;

[100;30m   [0m [91m                                    ~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(126,37): [91merror[0m TS2503: Cannot find namespace 'JQueryMousewheel'.

[100;30m131[0m             if ((<JQueryInputEventObject>evt.originalEvent).shiftKey || this.zoom_y == true) { // Zoom Y

[100;30m   [0m [91m                  ~~~~~~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(131,19): [91merror[0m TS2304: Cannot find name 'JQueryInputEventObject'.

[100;30m132[0m                 const position_y_px: number = (<JQueryMouseEventObject>evt.originalEvent).pageY - this.plot.offset().top;

[100;30m   [0m [91m                                                ~~~~~~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(132,49): [91merror[0m TS2304: Cannot find name 'JQueryMouseEventObject'.

[100;30m139[0m             } else if ((<JQueryInputEventObject>evt.originalEvent).altKey || this.zoom_x == true) { // Zoom X

[100;30m   [0m [91m                         ~~~~~~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(139,26): [91merror[0m TS2304: Cannot find name 'JQueryInputEventObject'.

[100;30m140[0m                 const position_x_px: number = (<JQueryMouseEventObject>evt.originalEvent).pageX - this.plot.offset().left;

[100;30m   [0m [91m                                                ~~~~~~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(140,49): [91merror[0m TS2304: Cannot find name 'JQueryMouseEventObject'.

[100;30m156[0m         $(window).bind('wheel', (event: JQueryEventObject) => {

[100;30m   [0m [91m        ~[0m

examples/pulse-generator/web/plot.ts(156,9): [91merror[0m TS2304: Cannot find name '$'.

[100;30m156[0m         $(window).bind('wheel', (event: JQueryEventObject) => {

[100;30m   [0m [91m                                        ~~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(156,41): [91merror[0m TS2304: Cannot find name 'JQueryEventObject'.

[100;30m157[0m             (<JQueryMousewheel.JQueryMousewheelEventObject>event).preventDefault();

[100;30m   [0m [91m              ~~~~~~~~~~~~~~~~[0m

examples/pulse-generator/web/plot.ts(157,15): [91merror[0m TS2503: Cannot find namespace 'JQueryMousewheel'.

[100;30m162[0m         $(window).unbind('wheel');

[100;30m   [0m [91m        ~[0m

examples/pulse-generator/web/plot.ts(162,9): [91merror[0m TS2304: Cannot find name '$'.

[100;30m230[0m             this.plot = $.plot(this.plot_placeholder, plt_data, this.options);

[100;30m   [0m [91m                        ~[0m

examples/pulse-generator/web/plot.ts(230,25): [91merror[0m TS2304: Cannot find name '$'.

[100;30m227[0m         offset(): JQueryCoordinates;

[100;30m   [0m [91m                  ~~~~~~~~~~~~~~~~~[0m

web/jquery.flot.d.ts(227,19): [91merror[0m TS2304: Cannot find name 'JQueryCoordinates'.

[100;30m263[0m     bind(method: string, handler: (eventObject: JQueryEventObject, ranges: jquery.flot.ranges) => any): JQuery

[100;30m   [0m [91m                                                ~~~~~~~~~~~~~~~~~[0m

web/jquery.flot.d.ts(263,49): [91merror[0m TS2304: Cannot find name 'JQueryEventObject'.

web/web.mk:20 : la recette pour la cible « tmp/pulse-generator/web/app.js » a échouée
```
I tried

```
sudo apt-get install npm
sudo npm install -g typescript
sudo npm install @types/node --save-dev
```
with no success

jeanminet, 2017-07-10:
Thanks.
I do not have the problem on my machine.
Could you try `$ sudo make setup` (or just `$ sudo make setup_web`) ?

llombard, 2017-07-11:
indeed, problem solved after the two commands.
Thanks!

closed 2017-07-11 by jeanminet