Koheron/koheron-sdk #416: File to patch option

CHV123, 2017-07-07:
Hi,

When I run "make" with the os option I get a query for a file to path:

# close_sw_design [current_sw_design]
# close_hw_design [current_hw_design]
INFO: [Common 17-206] Exiting hsi at Fri Jul  7 17:02:41 2017...
patch -d tmp/neutron_pulse_generator/os -p -0 < ./boards/red-pitaya/patches/devicetree.patch
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -rupN devicetree.old/pcw.dtsi devicetree/pcw.dtsi
|--- devicetree.old/pcw.dtsi
|+++ devicetree/pcw.dtsi
--------------------------
File to patch: 

When I run "make" with the all option it completes fine without errors.

I've also made a previous instrument but haven't had this option before. What file should I enter or is this a different problem?

Thanks,
Charlie

jeanminet, 2017-07-07:
Hi,
Are you running the latest version of the SDK ?
Did you try a `$ make clean` (or just `$ make clean_os`) ?
Jean

CHV123, 2017-07-10:
I tried a `make clean` and a `make clean_all` but I still got the option to patch after.

Charlie

jeanminet, 2017-07-10:
What version of Vivado are you using ?
Vivado 2017.1 introduced a change in the device tree structure.
The file `boards/red-pitaya/patches/devicetree.patch` is only compatible with device trees generated by versions >= 2017.1.
If are using Vivado 2016.4, you should use the tag `2016.4` of the SDK.
Jean

CHV123, 2017-07-11:
I'm using Vivado 2016.4. The command I am using is below, so I thought that I was using the 2016.4 tag? Is there something extra I need to add?

`sudo make VIVADO_VERSION=2016.4 CONFIG=instruments/neutron_pulse_generator/config.yml os`

Charlie

jeanminet, 2017-07-11:
Sorry, the tag is actually `vivado-2016.4` you should thus run `$ git checkout vivado-2016.4`.
Then you can just run `make CONFIG=instruments/neutron_pulse_generator/config.yml os`.

CHV123, 2017-07-11:
Thanks, it now works.

Charlie

closed 2017-07-11 by CHV123