Koheron/koheron-sdk #422: where is the right location for the gcc compiler?

jinsanko, 2017-08-21:
I just install your sdk and execute "make", below is my error message.
The arm-linux-gnueabihf-g++ should be somewhere under \opt\Xilinx\sdk\2017.2\gnu\aarch32\lin\gcc-arm-linux-gnueabi\bin if I follow Xilinx's procedure to install it in the root.

/usr/bin/arm-linux-gnueabihf-g++ -flto -c -Wall -Werror -Wextra -Wpedantic -Wfloat-equal -Wunused-macros -Wcast-qual -Wuseless-cast -Wlogical-op -Wdouble-promotion -Wformat -Wmissing-include-dirs -Wundef -Wcast-align -Wpacked -Wredundant-decls -Wvarargs -Wvector-operation-performance -Wswitch-default -Wuninitialized -Wshadow -Wzero-as-null-pointer-constant -Wmissing-declarations -Itmp/adc-dac/server -I./server/core -I. -I./server/context -I./server/drivers -Iexamples/adc-dac/ -DKOHERON_VERSION=0.16.0.f4cfba5 -MMD -MP -O3 -march=armv7-a -mcpu=cortex-a9 -mfpu=neon -mfloat-abi=hard -std=c++14 -pthread -o tmp/adc-dac/server/main.o server/core/main.cpp
bash: /usr/bin/arm-linux-gnueabihf-g++: No such file or directory
make[1]: *** [tmp/adc-dac/server/base64.o] Error 127

Thank for your help.

tvanderbruggen, 2017-08-21:
You can run
```
sudo make setup
```
to install all the required packages, including the cross-compiler.

jinsanko, 2017-08-22:
I have Vivado 2017.2 installed on Ubuntu 16.04, and pass "sudo make setup", but run into another error with "make CONFIG=examples/adc-dac/config.yml block_design"
The  message is
Traceback (most recent call last):
  File "./make.py", line 146, in <module>
    os.makedirs(output_dirname)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: 'tmp/adc-dac'
fpga/fpga.mk:14: recipe for target 'tmp/adc-dac/memory.yml' failed
make: *** [tmp/adc-dac/memory.yml] Error 1
jsk@ubuntu:~/jsk/Koheron-sdk$

Thank for your help.

jeanminet, 2017-08-22:
Hello,
It looks like a permission problem.
You can just delete the tmp directory (`$ sudo rm -rf tmp`) or change the permission (`$ sudo chmod +x tmp`).

jinsanko, 2017-08-23:
Thank. It works now and the scripts and Python code only supports 2017.1.

closed 2017-08-23 by jinsanko