Koheron/koheron-sdk #429: Fifo register allocation in c++ driver

CHV123, 2017-09-21:
Hi,

I've added a second AXI-4 Lite fifo to one of my instruments but I can't get any data out of it no matter how I arrange the ip in the block diagram. This leads me to think it's because I am using the same rdfd number for both fifo's in my c++ file. Would this be the case? If so, where can I find the relevant numbers for the second fifo?

Thanks for your help,
Charlie

CHV123, 2017-09-21:
I've found sort of my answer in the documentation, and as I understand it these numbers should be the same for both. However, this still doesn't solve my problem so I was wondering if you might have any suggestions?

jeanminet, 2017-09-21:
Hi Charlie,
As you said, the registers offsets should not change between the two FIFOs, only the base address is different.
Are you able to write and read the registers of the second FIFO ?
Jean

CHV123, 2017-09-21:
I don't seem to be writing or reading to the second fifo. I diagnosed this by looking at the output of the relevant AXI clock converter and the data did not change when connected to the second fifo, but it does change when connected to the first.

jeanminet, 2017-09-21:
Do you use the correct base address for the second FIFO ?
https://github.com/Koheron/koheron-sdk/blob/master/examples/pulse-generator/config.yml#L24
https://github.com/Koheron/koheron-sdk/blob/master/examples/pulse-generator/block_design.tcl#L105

CHV123, 2017-09-21:
The base address in the config file and block design match. I would have also thought that if the memory addressing was incorrect then I would have got a similar behaviour to previously when the addressing was incorrect.

jeanminet, 2017-09-21:
Then I don't know what to say.
You can send me your project by mail if you wish.

CHV123, 2017-09-22:
I've solved the problem for the moment by using the one fifo and an AXI width converter

closed 2017-09-22 by CHV123