Koheron/koheron-sdk #431: sudo make setup is not truly system-wide

jacklovell, 2017-10-09:
Hi

We have a development machine used for Koheron builds shared among several users. I ran into an issue with building the web app for the LED blinker the first time I tried it, despite my colleague having successfully built several Koheron instruments previously (and I'd successfully built the FPGA bitstream already). We traced the problem down to missing npm packages.

We'd assumed that "sudo make setup" would install everything system-wide (hence the sudo requirement), and this was mostly true. However, the setup_web target only installs typescript system-wide, and all the other npm packages are installed locally for a single user. Is this intentional, and if so could you please make it clearer in the documentation?

Thanks

Jack

jeanminet, 2017-10-09:
Hello Jack,
Thanks a lot for pointing the problem.
Does this commit 2130abc fix the issue ?
Jean

jacklovell, 2017-10-10:
Hi Jean

This change did result in the node packages being installed globally, but for some reason they're not found when trying to compile the web app:

```
jlovell@linfpga-3:~/koheron/koheron-sdk$ make VIVADO_VERSION=2016.4 CONFIG=~/jet_fpc/fpc_koheron/config.yml web
mkdir -p tmp/fpc_koheron/web
tsc --pretty --target ES3 --lib es6,dom --alwaysStrict web/koheron.ts /home/jlovell/jet_fpc/fpc_koheron/./web/led_blinker.ts /home/jlovell/jet_fpc/fpc_koheron/./web/app.ts /home/jlovell/jet_fpc/fpc_koheron/./web/control.ts --outFile tmp/fpc_koheron/web/app.js

101             let __WebSocket = require('websocket').w3cwebsocket;
                                  ~~~~~~~

web/koheron.ts(101,31): error TS2304: Cannot find name 'require'.

web/web.mk:20: recipe for target 'tmp/fpc_koheron/web/app.js' failed
make: *** [tmp/fpc_koheron/web/app.js] Error 2
make: *** Deleting file 'tmp/fpc_koheron/web/app.js'
```

However, the packages are definitely installed globally:
```
jlovell@linfpga-3:~/koheron/koheron-sdk$ npm list
/home/jlovell/koheron/koheron-sdk
└── (empty)

jlovell@linfpga-3:~/koheron/koheron-sdk$ npm list -g
/usr/local/lib
├── @types/jquery@2.0.46
├── @types/jquery-mousewheel@3.1.5
├── @types/node@8.0.33
├── typescript@2.5.3
└─┬ websocket@1.0.24
  ├─┬ debug@2.6.9
  │ └── ms@2.0.0
  ├── nan@2.7.0
  ├─┬ typedarray-to-buffer@3.1.2
  │ └── is-typedarray@1.0.0
  └── yaeti@0.0.6
```

Only by installing the packages locally (without the `--global` flag) can I get the web app to build.

jeanminet, 2017-10-10:
Would removing the two `--global` flags solve your problem ?

jacklovell, 2017-10-10:
I suspect so, but this would then result in the packages being installed only for me, which is counter-intuitive if sudo is required. In principle this is fine, but it would be useful to explicitly document that every user of a system needs to run at least this bit of the makefile (and ideally, users should not need to run sudo make if they're only installing these npm packages for themselves).

jeanminet, 2018-01-16:
Since #448 the setup command is `make setup` instead of `sudo make setup`

closed 2018-01-16 by jeanminet