Koheron/koheron-sdk #432: Default application on start up

grahamnaylorccfe, 2017-10-10:
I have developed and installed an instrument, which operates correctly after 'make ****** run', however when power cycling the unit reverts to the original instrument application. How can the default instrument by changed on power-up?
thanks
Graham

grahamnaylorccfe, 2017-10-10:
Charlie has just told me:
'You can set the default instrument in /usr/local/instruments/default '

grahamnaylorccfe, 2017-10-10:
Unfortunately that is not the whole story as the correct web page is installed, but not the bit file. The script install_instrument.sh is called from somewhere with the default bit file name. Where is this called from on start-up?

jeanminet, 2017-10-10:
Hello Graham,

You are right, there is a mistake in the start-up process.
At start-up, the instrument zip file defined in `/usr/local/instruments/default` is unzipped in `tmp/live-instrument`:
https://github.com/Koheron/koheron-sdk/blob/master/os/systemd/unzip-default-instrument.service
https://github.com/Koheron/koheron-sdk/blob/master/os/scripts/unzip_default_instrument.sh

When you start an instrument from the web interface or using `make run`, the script `install_instrument.sh` manages the installation of the bit file:

https://github.com/Koheron/koheron-sdk/blob/master/os/api/__init__.py#L34
https://github.com/Koheron/koheron-sdk/blob/master/os/api/install_instrument.sh

Unfortunately, the script is not called at start-up.
The temporary fix is to add the line:
`ExecStartPre=/bin/bash /usr/local/api/app/install_instrument.sh `basename $(cat /usr/local/instruments/default) .zip`
in `/etc/systemd/system/uwsgi.service` (see #433)

Note that you can also build your OS for your specific instrument
```
make CONFIG=myinstrument/config.yml os
sudo make CONFIG=myinstrument/config.yml image
```

Jean

grahamnaylorccfe, 2017-10-10:
I tried this, however I get some errors - .zip extensions seem to be added:

_______________________________________________________
root@koheron:~# /bin/bash /usr/local/api/app/install_instrument.sh $(cat /usr/local/instruments/default).zip
unzip:  cannot find or open /usr/local/instruments/AuroraTest.zip.zip.zip, /usr/local/instruments/AuroraTest.zip.zip.zip.zip or /usr/local/instruments/AuroraTest.zip.zip.zip.ZIP.
/usr/local/api/app/install_instrument.sh: line 11: /tmp/live-instrument/start.sh: No such file or directory
Load bitstream
/bin/cat: /tmp/live-instrument/AuroraTest.zip.zip.bit: No such file or directory
0
Restart koheron-server
Job for koheron-server.service failed because the control process exited with error code. See "systemctl status koheron-server.service" and "journalctl -xe" for details.
A dependency job for koheron-server-init.service failed. See 'journalctl -xe' for details.

_________________________________________________
For the moment I will add bash /usr/local/api/app/install_instrument.sh AuroraTest 
(where AuroraTest is the name of my instrument) to /usr/local/instruments/unzip_default_instruments.sh
It seems that /usr/local/api/app/__init__.py does not run?

jeanminet, 2017-10-10:
You need to run:
```
/bin/bash /usr/local/api/app/install_instrument.sh AuroraTest
```
As `/usr/local/instruments/defaults` contains `AuroraTest.zip`, you need to remove the zip extension:
```
/bin/bash /usr/local/api/app/install_instrument.sh `basename $(cat /usr/local/instruments/default) .zip`
```
The command you entered:
```
/bin/bash /usr/local/api/app/install_instrument.sh $(cat /usr/local/instruments/default).zip
```
adds another zip extension

closed 2017-10-13 by grahamnaylorccfe

tvanderbruggen, 2017-11-15:
The patch #437 should guarantee that the instrument specified in `/usr/local/instruments/default` (ex. `my_led_blinker.zip`) is run at startup. Of course the file `/usr/local/instruments/my_led_blinker.zip` must exist. You can thus define a default instrument which is not the one burned in the image.