Koheron/koheron-sdk #434: te0745 device crashing

grahamnaylorccfe, 2017-10-13:
I have been able to successfully run Aurora applications on the MYIR board, however when we built an image for the te0745, the first instrument worked correctly, however when we tried to add a second instrument with the make.... HOST=XXX run command the response was that it had been installed correctly, but the te0745 board crashed. The same operation on the MYIR did not give any problems. I then downloaded the image from your web site and tried building the stock example for the te0745, but simply changing the name to 'MonitorTest' so as not to confuse the device with another instrument of the same name and again the te0745 board crashed. Any ideas?

jeanminet, 2017-10-16:
> I then downloaded the image from your web site and tried building the stock example for the te0745, but simply changing the name to 'MonitorTest'

I followed your procedure and it worked.
Maybe it is related to the Vivado version (I used 2017.2).

grahamnaylorccfe, 2017-10-16:
I am using Vivado 2017.2 on a virtual machine running Ubuntu16.04.
I will try again (though I did try several times..). Strange.
Graham

jeanminet, 2017-10-16:
I don't understand.
Can you confirm that you use a SoM with the XC7Z035 Zynq?
Did you also change the name in the `config.yml` ?
Below is the switch configuration that I use:
[image]

grahamnaylorccfe, 2017-10-16:
Yes it is an XC7Z035 and the dip switches are as you show in your photo. The name is MonitorTemp in the config.yml file.

[cid:image001.png@01D346A1.FDA67560]

The device crashes immediately after the zip file is reported as successfully installed. However after power cycling the zip file is not in /usr/local/instruments as expected. If we copy it over manually and look at the available instruments the instrument MonitorTemp does appear, however clicking Start and then View causes it to crash again. We have tried manually unzipping the file and that seems to work correctly when targeting the live-instrument directory, but again crashes when we try and launch it. Perhaps something corrupt in the zip file?
I have attached it to see if you see the same problem with it.
Cheers
Graham

[Image removed by sender. 20171016_172059]<https://user-images.githubusercontent.com/1735094/31620552-d49f018e-b297-11e7-8bf6-e5fe70e05dfe.jpg>

This attachment was removed.

grahamnaylorccfe, 2017-10-17:
When I try and launch the instrument  manually, this is the transcript:

root@koheron:~# unzip -o /usr/local/instruments/MonitorTemp -d /tmp/live-ins                                        trument
Archive:  /usr/local/instruments/MonitorTemp.zip
  inflating: /tmp/live-instrument/MonitorTemp.bit
  inflating: /tmp/live-instrument/serverd
  inflating: /tmp/live-instrument/start.sh
  inflating: /tmp/live-instrument/_koheron.css
  inflating: /tmp/live-instrument/jquery.flot.js
  inflating: /tmp/live-instrument/jquery.flot.resize.js
  inflating: /tmp/live-instrument/jquery.flot.selection.js
  inflating: /tmp/live-instrument/jquery.flot.time.js
  inflating: /tmp/live-instrument/jquery.flot.axislabels.js
  inflating: /tmp/live-instrument/bootstrap.min.js
  inflating: /tmp/live-instrument/bootstrap.min.css
  inflating: /tmp/live-instrument/jquery.min.js
  inflating: /tmp/live-instrument/_koheron.svg
  inflating: /tmp/live-instrument/_koheron_logo.svg
  inflating: /tmp/live-instrument/kbird.ico
  inflating: /tmp/live-instrument/lato-v11-latin-400.woff2
  inflating: /tmp/live-instrument/lato-v11-latin-700.woff2
 extracting: /tmp/live-instrument/lato-v11-latin-900.woff2
  inflating: /tmp/live-instrument/glyphicons-halflings-regular.woff2
  inflating: /tmp/live-instrument/app.js
  inflating: /tmp/live-instrument/index.html
  inflating: /tmp/live-instrument/main.css
root@koheron:~# bash /usr/local/api/app/install_instrument.sh MonitorTemp
Archive:  /usr/local/instruments/MonitorTemp.zip
  inflating: /tmp/live-instrument/MonitorTemp.bit
  inflating: /tmp/live-instrument/serverd
  inflating: /tmp/live-instrument/start.sh
  inflating: /tmp/live-instrument/_koheron.css
  inflating: /tmp/live-instrument/jquery.flot.js
  inflating: /tmp/live-instrument/jquery.flot.resize.js
  inflating: /tmp/live-instrument/jquery.flot.selection.js
  inflating: /tmp/live-instrument/jquery.flot.time.js
  inflating: /tmp/live-instrument/jquery.flot.axislabels.js
  inflating: /tmp/live-instrument/bootstrap.min.js
  inflating: /tmp/live-instrument/bootstrap.min.css
  inflating: /tmp/live-instrument/jquery.min.js
  inflating: /tmp/live-instrument/_koheron.svg
  inflating: /tmp/live-instrument/_koheron_logo.svg
  inflating: /tmp/live-instrument/kbird.ico
  inflating: /tmp/live-instrument/lato-v11-latin-400.woff2
  inflating: /tmp/live-instrument/lato-v11-latin-700.woff2
 extracting: /tmp/live-instrument/lato-v11-latin-900.woff2
  inflating: /tmp/live-instrument/glyphicons-halflings-regular.woff2
  inflating: /tmp/live-instrument/app.js
  inflating: /tmp/live-instrument/index.html
  inflating: /tmp/live-instrument/main.css
Starting instrument
Clock initialization...
Load bitstream
0
Restart koheron-server
root@koheron:~# Write failed: Connection reset by peer

grahamnaylorccfe, 2017-10-17:
So the main thing to note seems to be that the bit file is not loaded. I try and do the same with the original te0745.bit file and the Load bitstream responds with 1 as expected. I do note that te0745.bit is 1749kB, whereas MonitorTemp.bit is 1036kB, yet this was nominally generated from the same source files. I will do another pull to make sure I have all your correct source files for this board.

jeanminet, 2017-10-17:
I change the zynq part from 7030 to 7035 a week ago. You may have compiled
the bitstream for the 7030.

grahamnaylorccfe, 2017-10-17:
Clang! Can you hear the penny drop? Yes that's the problem. Thanks for pointing that out! -durr....
G

closed 2017-10-17 by grahamnaylorccfe