Koheron/koheron-sdk #436: New board

grahamnaylorccfe, 2017-11-15:
Hi,
I am trying to create a new board definition for the snickerdoodle (with a view to bringing other boards online aswell). I have prepared the config and patch files from the hardware definition files, using diff etc. and try and test this by modifying one of the examples and then trying to build the os using:
make CONFIG=examples/snickerdoodle-led-blinker/config.yml os
Things seem to go OK for a while until it tries to get a non existent tar file from the Koheron repository:
".............
curl -L https://github.com/Koheron/u-boot-xlnx/archive/koheron-snickerdoodle-v2017.2.tar.gz -o tmp/u-boot-xlnx-koheron-snickerdoodle-v2017.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   150    0   150    0     0    268      0 --:--:-- --:--:-- --:--:--   268
100    15  100    15    0     0     13      0  0:00:01  0:00:01 --:--:--    13
"

Followed by the inevitable error:
"....
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
os/os.mk:67: recipe for target 'tmp/snickerdoodle-led-blinker/os/u-boot-xlnx-koheron-snickerdoodle-v2017.2' failed
"
Any idea of a way around this?
cheers
Graham

jeanminet, 2017-11-15:
Hi Graham,

We use a fork of u-boot-xlnx and a fork of linux-xlnx to manage the patches:
https://github.com/Koheron/u-boot-xlnx
https://github.com/Koheron/linux-xlnx

Each supported board (e.g. redpitaya) has a corresponding branch is each repository:
https://github.com/Koheron/u-boot-xlnx/tree/koheron-red-pitaya-v2017.2
https://github.com/Koheron/linux-xlnx/tree/koheron-red-pitaya-v2017.2

I hope that helps.
Jean

grahamnaylorccfe, 2017-11-16:
OK, so I fork a copy of the u-boot and linux repos and create branches for the Snickerdoodle (though the linux kernel seems to be unchanged for the different boards) and repoint the UBOOT_URL and LINUX_URL in os/os.mk to these. Does the archive tarball get created automatically by github when recovering the URL  ../archive/***.tar.gz file or should it be created using git archive?

jeanminet, 2017-11-16:
The archive tarball is created automatically.

grahamnaylorccfe, 2017-11-17:
I forked the repositories and added a branch for snickerdoodle-v2017.2 on the 2 repositories:
https://github.com/grahamnaylorccfe/u-boot-xlnx
https://github.com/grahamnaylorccfe/linux-xlnx

I point to these in the os/os.mk script
my u-boot tarball (with configs/zynq_snickerdoodle_defconfig etc edited) is recovered correctly into /tmp,but it fails again in the u-boot build:

________________________________
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
scripts/kconfig/Makefile:46: recipe for target 'silentoldconfig' failed
make[3]: *** [silentoldconfig] Error 1
Makefile:478: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 2
make[1]: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/uboot.release'. Stop.
___________________________________________________
Indeed the .config file was not created in tmp/snickerdoodle-led-blinker/os/u-boot-xlnx-koheron-snickerdoodle-v2017.2/

Is there something I have missed?
Graham

jeanminet, 2017-11-17:
Did you create an empty file named `zynq_snickerdoodle_defconfig` in `boards/snickerdoodle/patches` ?

grahamnaylorccfe, 2017-11-17:
Yep, that was the problem - thanks. Then I had to correct the config name from zynq-snickerdoodle to zynq_snickerdoodle. Now getting an error in the build of the kernel. I will chase that one on Monday...

grahamnaylorccfe, 2017-11-21:
Kernel build now sorted, so:
make CONFIG=examples/snickerdoodle-led-blinker/config.yml os
gets to [tmp/snickerdoodle-led-blinker/os/device-tree-xlnx-xilinx-v2017.2] OK
But when I do:
sudo make CONFIG=examples/snickerdoodle-led-blinker/config.yml image
I get a very weird error referring to an instrument I worked on months ago not referenced in this config:

_____________________________
cp: will not overwrite just-created '/tmp/ROOT.IOjAD71243/usr/local/instruments/AuroraTest.zip' with 'tmp/snickerdoodle-led-blinker/../TE0745_Aurora/AuroraTest.zip'
os/os.mk:48: recipe for target 'image' failed
make: *** [image] Error 1
________________________________

jeanminet, 2017-11-21:
I got this error once but I don't remember how I solved it...

This remind me that there is a problem because all available instruments (even ones built for other boards...) were copied to /usr/local/instruments.
This commit ce322c6293 only copies the default instrument defined by the CONFIG variable.

grahamnaylorccfe, 2017-11-22:
Great - that seems to have fixed it, thanks! Not yet checked whether snickerdoodle boots OK, but source on fork:
https://github.com/grahamnaylorccfe/koheron-sdk
with u-boot and kernal forks supporting snickerdoodle at:
https://github.com/grahamnaylorccfe/u-boot-xlnx
https://github.com/grahamnaylorccfe/linux-xlnx

grahamnaylorccfe, 2017-12-01:
I have built an SD card image and it boots OK, however the wifi  appears not to work (wpa_cli returns an error and no wifi port is found). It would seem that the nl80211 driver is not found (interface from wpa_supplicant to actual kernel drivers). I have looked through the linux-xlnx and u-boot-xlnx repos I created and those from KRTKL (makers of the snickerdoodle) and all the references to nl80211 are coherent and apparently enabled, however when I build with the Koheron SDK makefile I don't seem to get the wifi interface. Is this disabled in the build process by any chance?

jeanminet, 2017-12-01:
The wifi interface is not disabled in the build process but the script that builds the root file system (https://github.com/Koheron/koheron-sdk/blob/master/os/scripts/ubuntu-development.sh) does not install the tools (hostapd...) needed to make the wifi interface work.