Koheron/koheron-sdk #445: Clock wizard update from 2016.4 to 2017.2

CHV123, 2017-12-19:
I am still using vivado 2016.4 and it seems as if the clock wizard has been updated in the base block design tcl script. I just edited it manually back to version 5.3 within my files but would it be possible to generalise which version it calls in the future? 

Not really a problem as it was easy to fix but I thought I would give a heads-up just in case.

Thanks,
Charlie

jeanminet, 2018-01-11:
Hi Charlie,
I had no idea on how to implement it until I came upon this Xilinx forum thread:
https://forums.xilinx.com/t5/Vivado-TCL-Community/Get-IP-Catalog-ip-version/td-p/398415
Your problem should be fixed with #453.

CHV123, 2018-01-12:
Thanks Jean.

closed 2018-01-12 by CHV123