Koheron/koheron-sdk #459: python scripts for web using new page

CHV123, 2018-02-26:
Hi,

I'm trying to create a new web page called cgi-bin to run some python scripts. I've added a cgi-bin location to the nginx-server.conf file and then "include uwsgi_params" as seen in the location. I have also added a "root /usr/local/cgi-bin;" and created the corresponding directory and a hello world cgi-bin script.

This does not seem to work. Instead I get a 404 not found error when trying "koheron/cgi-bin" in a browser (the other instruments work fine in the browser). I have tried a few other things but either end up with a 404 or a 502 error. 

Do you have a quick start configuration for me so I can implement python scripts using "http://koheron/cgi-bin"?

I'm currently going through the digitalocean documentation but this is time sensitive so if you can quickly help out I figured it was worth asking.

Many thanks,
Charlie

jeanminet, 2018-02-26:
Hi Charlie,

Do you have a link for the digitalocean documentation ?

Thanks,
Jean

CHV123, 2018-02-26:
Hi Jean,

Yes, it's the link that appears in the nginx-server.conf file: https://www.digitalocean.com/community/tutorials/how-to-deploy-python-wsgi-applications-using-uwsgi-web-server-with-nginx

Charlie

jeanminet, 2018-02-26:
1. Create a systemd service in `/etc/systemd/system/my-uwsgi.service`:

```
[Unit]
Description=uWSGI
After=network.target

[Service]
ExecStartPre=/bin/mkdir -p /var/run/uwsgi
ExecStartPre=/bin/mkdir -p /var/log/uwsgi
ExecStart=/usr/local/bin/uwsgi --ini /etc/uwsgi/my-uwsgi.ini
Restart=always
KillSignal=SIGKILL
Type=notify
StandardError=syslog
NotifyAccess=all

[Install]
WantedBy=multi-user.target
```

2. Create a file `/etc/uwsgi/my-uwsgi.ini` for the configuration of uWSGI:

```ini
[uwsgi]
socket = /var/run/uwsgi/my-uwsgi.sock
chmod-socket = 666
chdir=/usr/local/my-api
wsgi-file = /usr/local/my-api/wsgi.py
; callable = app
master = true
; www-data uid/gid
;uid = www-data
;gid = www-data
uid = root
gid = root
die-on-term = true

; It is important to run the app in a single process, single thread environment
processes = 1
threads = 1

; Important when using subprocess during call
; http://stackoverflow.com/questions/14560010/nginx-uwsgi-and-flask-not-running-subprocess
close-on-exec = 1

; logger
; logger = file:/var/log/uwsgi/my-uwsgi.log
logger = internalservererror file:/var/log/uwsgi/my-uwsgi.log
log-route = internalservererror (HTTP/1.\d 500)
```

3. Create a file `/usr/local/my-api/wsgi.py`:

```py
from app import app as application

if __name__ == "__main__":
app.run(threaded=True, debug=True)
```

4. Create the file `/usr/local/my-api/__init__.py`

```py
# http://flask.pocoo.org/docs/0.12/quickstart/
import uwsgi
from flask import Flask
app = Flask(__name__)

@app.route('/')
def hello_world():
    return 'Hello, World!'
```

4. Add the following route in `/etc/nginx/sites-enabled/nginx-server.conf`:

```
location /my-api {
        uwsgi_pass              unix:/var/run/uwsgi/my-uwsgi.sock;
        uwsgi_read_timeout      60;

        # https://www.digitalocean.com/community/tutorials/how-to-deploy-python-wsgi-applications-using-uwsgi-web-server-with-nginx
        proxy_redirect          off;
        proxy_set_header        Host                 $host;
        proxy_set_header        X-Real-IP            $remote_addr;
        proxy_set_header        X-Forwarded-For      $proxy_add_x_forwarded_for;
        proxy_set_header        X-Forwarded-Proto    $http_x_forwarded_proto;
        proxy_set_header        X-Forwarded-Host     $server_name;

        # http://stackoverflow.com/questions/23751374/nginx-502-bad-gateway-when-uploading-files
        # define buffers, necessary for proper communication to prevent 502s
        proxy_buffer_size       128k;
        proxy_buffers 4         256k;
        proxy_busy_buffers_size 256k;

        include                 uwsgi_params;
}
```

5. Enable the systemd service using `$ systemctl enable my-uwsgi.service`

6. Reboot

7. Browse at  `http://ip-address/my-api`.

CHV123, 2018-03-01:
Hi Jean,

I've been struggling to get this running the past couple of days. I keep getting an internal server error. Was the above with the standard koheron 0.17 image?

In "journalctl -u my-uwsgi.service -f" I get "--- no python application found, check your startup logs for errors ---" but I am unsure where these log files are as "/var/log/uwsgi/" is empty.

The normal koheron site shows up fine using "http://ip-address".

Thanks for any help,
Charlie

jeanminet, 2018-03-01:
> Was the above with the standard koheron 0.17 image?

Yes

Can you try `$ journalctl -u my-uwsgi.service` instead to see the whole log?

CHV123, 2018-03-01:
So it seems to be having a problem importing the app module in wsgi.py:
`Traceback (most recent call last):`
`File "/usr/local/my-api/wsgi.py", line 1, in <module>`
`from app import app`
`ImportError: No module named app`
`unable to load app 0 (mountpoint='') (callable not found or import error)`

I've checked my __init__.py file and wsgi.py file and they are exactly as you have above (app.run line in wsgi,py is indented)

jeanminet, 2018-03-01:
My mistake !
`__init__.py` must be in a folder named `app` (see edit above)

CHV123, 2018-03-01:
Thanks Jean, the add-two now works.

closed 2018-03-01 by CHV123