Koheron/koheron-sdk #460: Memory allocation debugging

CHV123, 2018-03-13:
Hi,

My instrument driver (c++ bit) is throwing a std::bad_alloc exception. Do you know how I might go about debugging this on the RP?

Thanks,
Charlie

tvanderbruggen, 2018-03-14:
Hi,

When this happens I first try to guess the faulty container. If you recently introduced a `std::vector` or a `std::string` or something similar. I often get bad_alloc when I try to allocate a too large vector (for example mistake in power of 2 exponent for the size).

Then if I don't see an obvious problem, I usually try to find the faulty line using prints in the program. You can use either `ctx.log<INFO>("Here\n")` or `fprintf(stderr, "Here\n")`  in various places of your program and then check the result using `journalctl -u koheron-server`. Usually with some dichotomy it's quite fast to find the container that bad_alloc.

You can also use debugging tools such as gdb. You will have to change some compiler flags in [server.mk](https://github.com/Koheron/koheron-sdk/blob/master/server/server.mk): add `-g` to get debugging symbols and turn off optimizations `-O0`.

Hope that helps.

Thomas

CHV123, 2018-03-14:
It might help if I describe my problem a little better
:
My driver does return very large arrays (~16 million 32 bit values = 64 MB) but it manages this fine for the first 3 attempts. On the 4th it fails. If I use smaller arrays (12 million -> 5 successful, 8 million -> 10 successful) then it has more success, but still fails eventually.

 I have made sure to clear and re-allocate all my vectors within my driver and using print statements I am very sure that the problem is not within the instrument driver (I have put a try/catch clause and a print around the return statement of the array and this passes fine). 

Therefore I think the problem may be a memory leak within the koheron side. Do you know which files handle the array transfer from c++ to python? I'll try using gdb, how do I so that from the RP side?

Thanks,
Charlie

tvanderbruggen, 2018-03-14:
The server allocates a buffer in parallel to transfer the data and adds a packet header. We may have a memory leak there. I should be able to track it down if I have a minimal example of the crashing driver.

Would you be able to send a minimal version of your driver that reproduces the problem. A driver that would contain only the vectors, the associated management (clear, re-allocate, ...) and the function used to transfer the large array.

Also valgrind is probably a better tool than gdb to debug such memory problem.

CHV123, 2018-03-14:
[full_neutron_pulse.hpp.txt](https://github.com/Koheron/koheron-sdk/files/1810952/full_neutron_pulse.hpp.txt)
[minimal_neutron_pulse.hpp.txt](https://github.com/Koheron/koheron-sdk/files/1810954/minimal_neutron_pulse.hpp.txt)

I've attached a minimal version and the full version (just in case I missed something out).

Thank you for the help. I'll have a look at valgrind.

Charlie

tvanderbruggen, 2018-03-14:
I'm trying to reproduce the problem.
Can you send me a minimal python client to generate the same calling sequence.
Thanks.

CHV123, 2018-03-14:
To retrieve the buffer:
```
from koheron import command

class Neutron_Pulse(object):
    def __init__(self, client):
        self.client = client
        # self.n_pts = 16384
        self.n_pts = 2048
        self.fs = 125e6 # sampling frequency (Hz)
        self.adc = np.zeros((2, self.n_pts))
        self.dac = np.zeros((2, self.n_pts))
        @command()
        def get_adc_data(self):
              return self.client.recv_vector(dtype='uint32')
```

And then for the python client:

```
from neutron_pulse import Neutron_Pulse
from koheron import connect

host = os.getenv('HOST', 'RP ip address')
client = connect(host, name='neutron-pulse-generator')
driver = Neutron_Pulse(client)

dataArray = driver.get_adc_data()
```

dataArray will only be populated if there is data to collect so this goes in a while loop which waits for the device to be triggered:
```
while True:
    try:
        time.sleep(0.2)
        dataArray = driver.get_adc_data()
        if len(dataArray) > 1:
            break
    except KeyboardInterrupt:
       break
```
I have tested this while loop for interval time, and the number of successful acquisitions seems to be independent of interval time.

Thanks,
Charlie

tvanderbruggen, 2018-03-14:
The function `set_acquisition_length` should be called at some place no ?

tvanderbruggen, 2018-03-14:
Ah OK It is. I missed it.

It will be a bit difficult to reproduce without your bitstream.

I will try to generate a example downloding 64 MB from the pulse_generator example.

CHV123, 2018-03-14:
Sorry yes, you're right. I've edited the comment above and you will also need the acquisition length in the c++ as a public function:

```
 void set_acquisition_length(uint32_t acquisition_length) {
    ctx.log<INFO>("Setting acquisition length");
    ctl.write<reg::acquisition_length>(acquisition_length);
    samples2collect = (acquisition_length) * 4;
  }
```

tvanderbruggen, 2018-03-14:
Here is a driver that runs on the pulse_generator bitstream:
```c++
/// Pulse driver
///
/// (c) Koheron

#ifndef __DRIVERS_PULSE_HPP__
#define __DRIVERS_PULSE_HPP__

#include <atomic>
#include <thread>
#include <chrono>

#include <context.hpp>

// http://www.xilinx.com/support/documentation/ip_documentation/axi_fifo_mm_s/v4_1/pg080-axi-fifo-mm-s.pdf
namespace Fifo_regs {
    constexpr uint32_t rdfr = 0x18;
    constexpr uint32_t rdfo = 0x1C;
    constexpr uint32_t rdfd = 0x20;
    constexpr uint32_t rlr = 0x24;
}

constexpr uint32_t fifo_buff_size = 16777216;

class Pulse
{
  public:
    Pulse(Context& ctx)
    : ctl(ctx.mm.get<mem::control>())
    , sts(ctx.mm.get<mem::status>())
    , adc_fifo_map(ctx.mm.get<mem::adc_fifo>())
    ,adc_data(fifo_buff_size)
    {
        start_fifo_acquisition();
    }

    // Adc FIFO

    uint32_t get_fifo_occupancy() {
        return adc_fifo_map.read<Fifo_regs::rdfo>();
    }

    void reset_fifo() {
        adc_fifo_map.write<Fifo_regs::rdfr>(0x000000A5);
    }

    uint32_t read_fifo() {
        return adc_fifo_map.read<Fifo_regs::rdfd>();
    }

    uint32_t get_fifo_length() {
        return (adc_fifo_map.read<Fifo_regs::rlr>() & 0x3FFFFF) >> 2;
    }

    void wait_for(uint32_t n_pts) {
        do {} while (get_fifo_length() < n_pts);
    }

    const auto& get_adc_data() {
        return adc_data;
    }

    void start_fifo_acquisition();

  private:
    Memory<mem::control>& ctl;
    Memory<mem::status>& sts;
    Memory<mem::adc_fifo>& adc_fifo_map;

    uint32_t pulse_width;
    uint32_t pulse_period;

    std::vector<uint32_t> adc_data;

    std::atomic<bool> fifo_acquisition_started{false};
    std::atomic<uint32_t> fifo_buff_idx{0};

    std::thread fifo_thread;
    void fifo_acquisition_thread();

};

inline void Pulse::start_fifo_acquisition() {
    if (! fifo_acquisition_started) {
        fifo_thread = std::thread{&Pulse::fifo_acquisition_thread, this};
        fifo_thread.detach();
    }
}

inline void Pulse::fifo_acquisition_thread()
{
    constexpr auto fifo_sleep_for = std::chrono::microseconds(5000);

    fifo_acquisition_started = true;

    while (fifo_acquisition_started) {
        const uint32_t n_pts = get_fifo_length();

        for (size_t i = 0; i < n_pts; i++) {
            adc_data[fifo_buff_idx] = read_fifo();
            fifo_buff_idx = (fifo_buff_idx + 1) % fifo_buff_size;
        }
        std::this_thread::sleep_for(fifo_sleep_for);
    }
}

#endif // __DRIVERS_PULSE_HPP__
```

```py
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import numpy as np
import matplotlib
matplotlib.use('GTKAgg')
from matplotlib import pyplot as plt
import os
import time

from koheron import connect, command

class Pulse(object):
    def __init__(self, client):
        self.client = client

    @command()
    def get_adc_data(self):
        return self.client.recv_vector(dtype='uint32')

host = os.getenv('HOST', '192.168.1.15')
client = connect(host, name='fix-mem-leak')
driver = Pulse(client)

while True:
    data = driver.get_adc_data()
    print data.size
```
I run it for several minutes and downloaded many 64MB buffers. I don't observe a memory leak.

A single buffer `adc_data` is used. It is allocated to 64MB (2**16 points) in the constructor. This buffer is filled by the FIFO thread and send directly to the python client.

In your example the buffering scheme is fairly complicated. You have three buffers and a two copies to transfer between them. So you are using 3 times the required amount of memory and as a result it is a significant fraction of the system memory (512 MB).

Also the reserve function will only increase the allocated memory of the vector. It will never decrease unless you explicitly call shrink_to_fit. But usually you don't need to call those functions, it is easier to allocate upfront (in the constructor) the required memory.

CHV123, 2018-03-14:
I will give this a go, but I think when I first tried the standard driver it resulted in dropped samples. I also need to be able to poll the buffer to check when all the data is there so that it can be retrieved which caused problems as it resulted in simultaneously writing to and requesting the length of the adc_data vector. I cannot instantiate a fifo big enough to hold the entire data set either as there is not enough available memory within my logic design.

CHV123, 2018-03-14:
How would I go about using valgrind to diagnose my problem?

Thanks,
Charlie

tvanderbruggen, 2018-03-14:
To access simultaneously the length of the adc_data buffer while writing to it you can for example access safely the variable `fifo_buff_idx` in my example since it is `std::atomic`. Else you can use `lock_guard` to safely to protect region accessing vector data from different threads.

CHV123, 2018-03-14:
I've tried to implement your above method using 
```
uint32_t get_buffer_length() {
    return fifo_buff_idx;
  }
```
To poll the buffer size but even without the sleep this is too slow and results in dropped samples.

I have adapted my original function to only use one vector, and I can now get 5 successful 16 million sample acquisitions before a crash (as opposed to 3). I would like to use valgrind to see where the memory leak may be coming from. How do I go about this on the RP?

Thanks,
Charlie

CHV123, 2018-03-15:
Using top I've found that the process serverd is not clearing its memory after each acquisition. Is this helpful?

Thanks,
Charlie

tvanderbruggen, 2018-03-15:
There is no reason for it to clear its memory. It's better to allocate upfront a large chunk of memory than deallocating and reallocating memory after each acquisition (it takes time and leads to memory fragmentation). Calling reserve will only increase the capacity of the vector.

If you reserve in constructor the whole memory required for your buffer(s) and then you don't call reserve at all elsewhere then no memory allocation will be done outside the constructor. A bad_alloc could only happen at driver initialization if not enough memory is available on the system and never after.

I don't have a clear procedure for running valgrind on the RedP. I might try to define one but I don't have much time right now. In principle it should enough to install valgrind on the ReP on to called it according to http://valgrind.org/docs/manual/quick-start.html passing the serverd executable in argument.

CHV123, 2018-03-15:
So I think that I've found my problem. Every time that I was acquiring data I was re-running my python acquisition script. This caused a new instance of serverd to open on the RP as opposed to using the already running instance, meaning that serverd ended up not having any memory to allocate to, causing the bad_alloc().

When running the python script I can't see any options on whether to run a new serverd instance or to use the one already in place. Maybe there should be something to close the serverd instance when the client disconnects from the RP? Or a way of not opening a new instance and instead using the one already running?

Thank you for your help,
Charlie

closed 2018-03-15 by CHV123

tvanderbruggen, 2018-03-15:
This is weird. Only one server can be running at the time since it is restarted at each instrument installation (see [install_instrument.sh](https://github.com/Koheron/koheron-sdk/blob/master/os/api/install_instrument.sh)).

When using the `connect` function there is an option to restart the instrument or not at each connection. But in any case only one serverd instance should be running at the time.

I don't think this is the explanation for your problem.

CHV123, 2018-03-15:
You may be right that it's not another instance but each time a new connection is made with the python script the memory usage of serverd goes up by ~62MB. However, when I use a web that keeps the connection open the memory usage stays constant.

For the moment I can pull as much data as I want while running through the web app.

tvanderbruggen, 2018-03-15:
You're right. For each connection the server will create a new dedicated communication buffer. This buffer will grow up two the size of the biggest vector transfer by the connection. Their is no shrinking mechanism implemented for now(not obvious to determine under which condition it should shrink).

If you required several connections to transfer 64Mb blobs of data you will quickly exhaust the RedP memory. You should have a single connection dedicated to the transfer of this large buffer.

CHV123, 2018-03-15:
What would I put in the python script to tell it to use the already opened buffer as opposed to opening a new one? I could see in journctl that once the script had stopped running the connection had closed, but serverd must be keeping the buffer open. Is there a way to check for an already running buffer and connecting to that as opposed to opening a new one?

tvanderbruggen, 2018-03-15:
Once a connection is closed the buffer associated to it is deleted and it's memory is freed.
So opening and closing connections should not be an issue as long as only one connection is transferring the large buffer at the time.