Koheron/koheron-sdk #465: symlink loop with nodejs

CHV123, 2018-04-25:
Hi,

In trying to sort out a json error I was getting, i ended up removing and re-installing nodejs and running the make setup_web command. I ended up with a "can't execute 'node': Too many levels of symbolic links" error. This occurs because setup_web deletes `/usr/bin/node` and then creates a symlink from  `/usr/bin/nodejs` to `/usr/bin/node` , but `/usr/bin/nodejs` is a symlink from `/etc/alternatives/nodejs`, which is a symlink to the (previously deleted) ` /usr/bin/node` file. There must be something wrong with my nodejs installation right? Do you have any suggestions?

Thanks,
Charlie

jeanminet, 2018-05-03:
Hi Charlie,

The last problem `ValueError: No JSON object could be decoded` is now fixed with 1de12a8b.

Regarding the Node installation, I tried to fix it with c240ba15.
You can try to remove Node with `sudo apt-get remove nodejs`
and then run `make setup_web` 

Jean

CHV123, 2018-05-03:
Hi Jean,

Thanks for this, the ValueError message seems to be gone.

I removed nodejs but when running `make setup_web` I failed with:
```
The following packages have unmet dependencies:
npm : Depends: nodejs but it is not going to be installed
       Depends: node-abbrev (>= 1.0.4) but it is not going to be installed
       Depends: node-ansi (>= 0.3.0-2) but it is not going to be installed
       Depends: node-ansi-color-table but it is not going to be installed
       Depends: node-archy but it is not going to be installed
       Depends: node-block-stream but it is not going to be installed
       Depends: node-fstream (>= 0.1.22) but it is not going to be installed
       Depends: node-fstream-ignore but it is not going to be installed
       Depends: node-github-url-from-git but it is not going to be installed
       Depends: node-glob (>= 3.1.21) but it is not going to be installed
       Depends: node-graceful-fs (>= 2.0.0) but it is not going to be installed
       Depends: node-inherits but it is not going to be installed
       Depends: node-ini (>= 1.1.0) but it is not going to be installed
       Depends: node-lockfile but it is not going to be installed
       Depends: node-lru-cache (>= 2.3.0) but it is not going to be installed
       Depends: node-minimatch (>= 0.2.11) but it is not going to be installed
       Depends: node-mkdirp (>= 0.3.3) but it is not going to be installed
       Depends: node-gyp (>= 0.10.9) but it is not going to be installed
       Depends: node-nopt (>= 3.0.1) but it is not going to be installed
       Depends: node-npmlog but it is not going to be installed
       Depends: node-once but it is not going to be installed
       Depends: node-osenv but it is not going to be installed
       Depends: node-read but it is not going to be installed
       Depends: node-read-package-json (>= 1.1.0) but it is not going to be installed
       Depends: node-request (>= 2.25.0) but it is not going to be installed
       Depends: node-retry but it is not going to be installed
       Depends: node-rimraf (>= 2.2.2) but it is not going to be installed
       Depends: node-semver (>= 2.1.0) but it is not going to be installed
       Depends: node-sha but it is not going to be installed
       Depends: node-slide but it is not going to be installed
       Depends: node-tar (>= 0.1.18) but it is not going to be installed
       Depends: node-underscore but it is not going to be installed
       Depends: node-which but it is not going to be installed
```

After some [research](https://askubuntu.com/questions/462337/couldnt-install-npm-on-ubuntu?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa) it looks like npm is already installed (`npm -v ` returns 6.0.0) and I can use the following two npm commands without `sudo apt-get install -y npm`.

Thanks,
Charlie