Koheron/koheron-sdk #475: Accessing Red Pitaya IO Pins

Rondin119, 2018-08-24:
Hello,

I am in the process of developing an RP to run a gas control system for an experiment I am running soon. In order to do this I need to access a large number of the IO pins on the RP Board (16 outputs and 2 inputs) all of which only need to go low or hi. I have tried to add them via the expansion connector, however vivado will give me an error: 

[Place 30-415] IO Placement failed due to overutilization. This design contains 216 I/O ports
 while the target  device: 7z010 package: clg400, contains only 38 available user I/O. The target device has 230 usable I/O pins of which 192 are already occupied by user-locked I/Os.
 To rectify this issue:
 1. Ensure you are targeting the correct device and package.  Select a larger device or different package if necessary.
 2. Check the top-level ports of the design to ensure the correct number of ports are specified.
 3. Consider design changes to reduce the number of I/Os necessary.

I understand this is due to the fact that each physical pin normally requires 12 ports, my question is if there is a way to make the pins accept an STD_LOGIC input (or a single port) instead of a 12 bit Logic_Vector? I only need to make the pins go hi and low so I don't care about the range.

Thanks,

William

jeanminet, 2018-08-27:
Hello William,
Could you share your .xdc constraints file?
Thanks,
Jean

grahamnaylorccfe, 2018-09-12:
This would be for controlling a stellerator then?
Have you tried starting with one of the standard designs? Not sure why you are using 12 bit vectors.
G

Rondin119, 2018-09-18:
Hi Graham,

Indeed it is, well a specific diagnostic on it. Looking back at the standard designs helped me find the answer, thank you. I don't know why I didn't think of doing that earlier. 

William

Rondin119, 2018-09-18:
Hi Jean,

Thanks for your reply, I have reworked the design to not use those pins.

Thanks again,
William McCarthy

closed 2018-09-18 by Rondin119