Koheron/koheron-sdk #478: How do I create an SDK project?

wally900809, 2018-09-10:
Hi
I use the alpha250 board to implement the fft function.
I uploaded an FFT example for alpha250 to SD card using Ubuntu and vivado 2017.2. I have also run a web page using make make HOST=192.~ run.
I would like to modify a few things in this example.
There are several questions to that end.

1. In the FFT example, does the SDK project file or Folder executable in the Xilinx SDK program exist?
If the answer to Question 1 is 'not exist', where do I modify the application of zynq board? For example, does not main.c or main.cpp work for the FFT example?

2. I would like to send only certain data values using Ethernet communication without displaying the FFT results using the website. Which part of the example should I modify for this behavior?

3. Can I upload images to qspi using jtag without SD card?

Thank you for reading this question for beginners.

jeanminet, 2018-09-14:
Hi,

1. The FFT example uses Linux.
It is not a bare-metal application compiled with Xilinx SDK.
The source code of the FFT example can be found here: https://github.com/Koheron/koheron-sdk/tree/master/examples/alpha250/fft

2. You can try editing the [`fft.hpp`](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/fft/fft.hpp) file. Each function (.e.g. `read_psd()`, `set_input_channel()`...) can be accessed from a Python script running on your computer (see examples here: https://github.com/Koheron/koheron-sdk/tree/master/examples/alpha250/fft/python).

3. The BOOT MODE switch allows to choose between boot using SD card (OFF) and QSPI (ON).
Note that we do not provide support for the QSPI mode.

Jean

wally900809, 2018-09-19:
Thank you for answer. I want to ask one more thing. Running the example will create a tmp folder.
What does the .hpp and .cpp files in the "tmp / examples / alpha250 / fft / server" path do?

Also, when you run the FFT example on the board, I would appreciate briefly describing in what order the FPGA Logic and os, server, web, and drivers files specified in the makefile and config.yml file are executed for FFT operation.

Or can you tell us about the sequence of operations between "logic <-> server <-> web"?

jeanminet, 2018-09-19:
The server is built using the (`server.mk`)[https://github.com/Koheron/koheron-sdk/blob/master/server/server.mk] Makefile. 
Each driver defined in the config.yml (e.g. `fft.hpp`) is analyzed by a Python script that generates a file called `interface_fft.hpp` in the `tmp/examples/alpha250/fft/server`. This file allows to map each function of the driver to a command in the server.

The Makefile does not really define a sequence of operations but a list of dependencies.
For instance, when you call the `$ make run` command,

```Makefile
.PHONY: run
run: $(INSTRUMENT_ZIP)
	curl -v -F $(NAME).zip=@$(INSTRUMENT_ZIP) http://$(HOST)/api/instruments/upload
	curl http://$(HOST)/api/instruments/run/$(NAME)
	@echo
```

The Makefile sees that it needs the instrument zip file:

```Makefile
# Zip file that contains all the files needed to run the instrument:
INSTRUMENT_ZIP := $(TMP_PROJECT_PATH)/$(NAME).zip
$(INSTRUMENT_ZIP): server $(BITSTREAM) $(START_SH) web $(VERSION_FILE)
	zip --junk-paths $(INSTRUMENT_ZIP) $(BITSTREAM) $(SERVER) $(START_SH) $(WEB_ASSETS) $(VERSION_FILE)
	@echo [$@] OK
```

So it needs to build the server, the bitstream...

``` Makefile
$(BITSTREAM): $(TMP_FPGA_PATH)/$(NAME).xpr | $(TMP_FPGA_PATH)
	$(VIVADO_BATCH) -source $(FPGA_PATH)/vivado/bitstream.tcl -tclargs $< $@
	@echo [$@] OK
```

And so on...
I'm not sure this is the information you need.
Let me know.

wally900809, 2018-10-24:
Thanks Jean. After the answer, I tried many things. I have some questions.

The communication method between server and client is TCP / IP. I want to change TCP / IP to UDP. First we looked at the files in koherom-sdk-master / server / core.
Simply change the socket function variable from SOCK_STREAM to SOCK_DGRAM in the listening_channel.cpp file and build it. Then, on the board's Linux OS, FAILED is displayed on the serial monitor as Failed to start Koheron TCP / Websocket server.
Which file should be modified to change the communication method from TCP to UDP? Or are there any references or examples?