Koheron/koheron-sdk #480: Xilinx SDK & server

rsarwar87, 2018-09-26:
Hi,

Firstly thank you for sharing this tool. the process is a bit different to what I am used to with Altera boards, but I really like the Koheron-sdk features.

I had a couple of question regarding the use of Xilinx SDK and the remote server interface.
1. Is there a way I could possibly execute an elf, generated from Xilinx SDK, in the Koheron system? When I tried I get an error. I'll edit the post later in the day with the exact error message.

2. Secondly, I really like the server interface available with the K-SDK that forms the instrument drivers, and I was wondering if there was a C/C++ interface (instead of python) which can be used to build a remote application (hosted in a linux laptop/desktop) to communicate with the instruments.

Br
Rashed

jeanminet, 2018-09-26:
Hi Rashed,

Thanks. Regarding the C++ client, I will add a basic example in the next few days.

What do you mean by executing an elf generated by XIlinx SDK?

Best,
Jean

rsarwar87, 2018-09-26:
Hi Jean

Thank you. Regarding the elf, I wanted know if it is possible to execute elf file generated by Xilinx SDK (Vivado: File>launch SDK) using system_wrapper_plaform (Vivado: File>export>Export Hardware). I tried a simple code and it returned "Illegal Instruction" when executed.

```
#include <stdio.h>
#include "platform.h"
#include "xil_printf.h"
int main()
{
    print("Hello World \n\r");
    return 0;
}

```
I am on: koheron 4.9.0-xilinx  Mon Sep 3 20:25:34

jeanminet, 2018-09-28:
I don't think you can execute a bare-metal application on Linux.
You cannot directly access memory on Linux...

However, I tried compiling an "Hello world" Linux application with Xilinx SDK.
The generated elf executes correctly.

jeanminet, 2018-09-28:
Regarding the C++ client, I added an example with #483.

rsarwar87, 2018-09-28:
Hi, 

Thank you for the C++ client, much appreciated. 

Just out of curiosity, any plans to do the same for Altera SoCs maybe? Terasic has some interesting Cyclone V SoCs that could be interesting when paired with your SDK? 

Br 
Rd

closed 2018-10-01 by rsarwar87