Koheron/koheron-sdk #481: Breaks on systems where the `python` executable points to Python 3

jacklovell, 2018-09-27:
There are a number of areas in the Python scripts in the SDK which are incompatible with Python 3. This means that on a system where the `python` executable is Python 3, the `make` commands to build and instrument will not run. This can happen for example if a user has installed Anaconda3, and prepends the Anaconda BIN path to `$PATH`.

Making the code base Python 3 compatible is (presumably) non-trivial. But a simpler interim fix could be implemented: change the `python` calls in the various makefiles to `python2` calls. All systems I'm aware of still have a `python2` executable, so this will enforce the use of Python 2, as long as it's installed.

I've managed to get this to work by modifying only 2 files: `python/python.mk` and `Makefile`, but there may be others I've missed.

jeanminet, 2018-09-28:
Thanks for reporting the issue.
It should now be fixed with #482.

jacklovell, 2018-09-28:
Thanks, that fix seems to do the trick when I pass PYTHON=python2 to the make commands.

closed 2018-09-28 by jacklovell

jeanminet, 2018-09-28:
Could you try with `PYTHON=python3` ?
I had no problem running the make command with python3 on my setup.

jacklovell, 2018-09-28:
Yep, that works too, once I'd installed the Python dependencies for Python 3 (`make setup` had been run with Python 2 initially).

Rather annoyingly, Anaconda is inconsistent in that it creates both `python` and `python3` executables, but only `pip` not `pip3`. So I'd still need to use `make PIP=pip setup` if I wanted to install all the dependencies on a clean OS in an Anaconda install, rather than the system-wide Python 3 installation. I ended up just running `pip install -r requirements.txt` and `pip install ./python/` manually though, which worked well enough.

I think it's fair to say that's Anaconda's problem to fix rather than yours. Particularly as there is already a workaround.