Koheron/koheron-sdk #488: Server crashing

grahamnaylormrc, 2018-11-06:
I have developed an application for the Alpha250 board to measure the phase of an RF signal on one of the ADC channels (frequency generated by the clock chip). This worked nicely with the version 0.17. I use a simple python program to configure the device and pull data off, but now I have updated to version 0.19 the instrument re-builds loads and runs OK, but when I access the registers with the python program the ARM seems to crash and is no longer responsive over the network (can no longer ping it).

grahamnaylormrc, 2018-11-06:
I have added a clock configuration to the clock-generator.hpp file (increasing the number of configs to 3), but I think there is some issue there as the problem arises after setting the sampling frequency and reference clock. Will look further..

jeanminet, 2018-11-06:
Thank you for reporting the issue.
If the AXI bus is being accessed while it is not clocked, it crashes the ARM.
The clock generator should always output a clock.
I will also look into this...

jeanminet, 2018-11-06:
The issue was already present in version 0.17...
It should be fixed with a624aee.

grahamnaylormrc, 2018-11-06:
I think the issue was a little more than that - though thanks for looking into it. The FPGA would start  with the FFT instrument (so the clock chip should already have been set up). The gremlin that crept in at some point I believe was my error in the addition of the 3rd clock config in clock-generator.hpp. It was OK, but then at some point it became this:

    constexpr size_t num_configs = 3;
    constexpr size_t num_params = 10;

    // Sampling frequency 200 MHz (f_vco = 2400 MHz)
    constexpr std::array<uint32_t, num_params> fs_200MHz = {
        2,   // PLL2_P
        12,  // PLL2_N
        2,   // PLL2_R
        240, // CLKout0_DIV (CLKOUT)
        12,  // CLKout1_DIV (ADC clock)
        12,  // CLKout2_DIV (DAC clock)
        12,  // CLKout3_DIV (FPGA clock)
        240, // CLKout4_DIV (EXP_CLK0 clock)
        240, // CLKout5_DIV (EXP_CLK1 clock)
        0    // MMCM phase shift
    };

    // Sampling frequency 250 MHz (f_vco = 2500 MHz)
    constexpr std::array<uint32_t, num_params> fs_250MHz = {
        5,   // PLL2_P
        5,   // PLL2_N
        2,   // PLL2_R
        250, // CLKout0_DIV (CLKOUT)
        10,  // CLKout1_DIV (ADC clock)
        10,  // CLKout2_DIV (DAC clock)
        10,  // CLKout3_DIV (FPGA clock)
        250, // CLKout4_DIV (EXP_CLK0 clock)
        250, // CLKout5_DIV (EXP_CLK1 clock)
        56   // MMCM phase shift
    };

    // Sampling frequency 240 MHz (f_vco = 2400 MHz)
   // Fvco= 100*PLL2_N*PLL2_P/PLL2_R    -> PLL2_P is the prescalar and can be 2-8, whereas PLL2_N can be 1-262143 and PLL2_R can be 1-4095
    constexpr std::array<uint32_t, num_params> for_pm = {
        2,   // PLL2_P
        12,   // PLL2_N
        2,   // PLL2_R
        12, // CLKout0_DIV (CLKOUT) - this goes to the SMA
        10,  // CLKout1_DIV (ADC clock)
        10,  // CLKout2_DIV (DAC clock)
        10,  // CLKout3_DIV (FPGA clock)
        240, // CLKout4_DIV (EXP_CLK0 clock)
        240, // CLKout5_DIV (EXP_CLK1 clock)
        56   // MMCM phase shift
    };

    constexpr std::array<std::array<uint32_t, num_params>, num_configs> configs = {fs_200MHz, fs_250MHz};
}

Note the lack of entry in the last line for 'for_pm' ,and  should of course have been:
    constexpr std::array<std::array<uint32_t, num_params>, num_configs> configs = {fs_200MHz, fs_250MHz, for_pm};

durr
Graham

closed 2018-11-06 by grahamnaylormrc