Koheron/koheron-sdk #490: Implementation Simulation of an IP Packager generated IP

rsarwar87, 2018-11-13:
Hi,

This is probably not an issue with the SDK, but me being relatively limited experience with vivado.

I have a custom IP that I created using IP Packager. The  it is kept in ./instruments/<name>/<IP name>.
The vivado package files are generated in ./tmp/instruments/<name>

The block_design was modified to reflect the IP. 
`set_property "ip_repo_paths" "[concat [get_property ip_repo_paths [current_project]] [file normalize "/wkspace/fpga/koheron-workspace/examples/red-pitaya/rp-psa-dual/psa_core_1.1"]]" "[curr
ent_project]"                                                                                                                                                                                
update_ip_catalog -rebuild                                                                                                                                                                   
                                                                                                                                                                                             
cell user.org:user:psa_core:1.1 psa_core_0 {                                                                                                                                                 
                                                                                                                                                                                             
} {                                                                                                                                                                                          
    CLK adc_dac/adc_clk                                                                                                                                                                      
    RST proc_sys_reset_adc_clk/peripheral_aresetn                                                                                                                                            
    ADC0_IN adc_dac/adc1                                                                                                                                                                     
    ADC1_IN adc_dac/adc2                                                                                                                                                                     
    CONFIG0 [get_concat_pin [list [ctl_pin psa0_high] [ctl_pin psa0_low]]]                                                                                                                   
    CONFIG1 [get_concat_pin [list [ctl_pin psa1_high] [ctl_pin psa1_low]]]                                                                                                                   
    TIMEOUT [sts_pin reg1]                                                                                                                                                                   
    bram_psa axi_bram_ctrl_psa/BRAM_PORTA
    dma_stream dma_axis_clock_converter/S_AXIS
    psa_fifo psa_axis_clock_converter/S_AXIS
}`

I then open the vivado project, create a testbench file which instantiates the IP and pass in some stimulus. However, when running the simulation, I end up with an error message:
`WARNING: [VRFC 10-122] psa_core_top remains a black-box since it has no binding entity [/wkspace/fpga/koheron-workspace/examples/red-pitaya/rp-psa-dual/psa_core_1.1/src/tb_fulltest.vhd:229]`
OOC build is not implemented.

Not sure if this is me not fully following the vivado specific step or SDK specific process. I was wondering what the correct way to add a reference to an IP is.

jeanminet, 2018-11-18:
Hi,
I must admit I don't know what is going on.
Maybe this thread could help:
https://www.xilinx.com/support/answers/64349.html

rsarwar87, 2018-11-18:
Hi

Thanks, That did solve the problem. Might I ask if you were able to recreate the issue I was mentioning.

Br
Rashed

closed 2019-08-14 by rsarwar87