Koheron/koheron-sdk #493: Precision ADC and DAC on alpha 250

grahamnaylormrc, 2019-01-31:
Do you have any example python scripts accessing the precision ADC and DAC?
thanks
Graham

jeanminet, 2019-01-31:
Below is a simple example:

```python
import time
import os
import numpy as np
from koheron import connect, Alpha250

host = os.getenv('HOST', '192.168.1.100')
client = connect(host, 'fft', restart=False)

driver = Alpha250(client)

dac_channel = 0
adc_channel = 0

v_min = 0.0 # V
v_max = 1.0 # V

for dac_voltage in np.linspace(v_min,v_max,1000):
    time.sleep(0.1)
    driver.set_precision_dac_volts(dac_channel, dac_voltage)
    adc_values = driver.get_precision_adc_values()
    print("ADC{} = {:.5f} V".format(adc_channel, adc_values[adc_channel]))
```

Note that this only works if you compiled your instrument with the [precision-adc](https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250/drivers/precision-adc.hpp) and [precision-dac](https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250/drivers/precision-dac.hpp) drivers. The Alpha250 driver is included in the koheron Python package, you can find the source code [here](https://github.com/Koheron/koheron-sdk/blob/master/python/koheron/alpha250.py)

Jean

tvanderbruggen, 2019-02-01:
Hi Graham,

I reproduced your setup using a pair of 220 Ohm as a voltage divider on VREF producing 1.25 V (I put no 100 u cap). I only connected the 1.25 V to a single ADC input (N pin) and I connected a single DAC output to the the same ADC P pin.

Here is what I observe:

```
ADC0 = -1.24076 (-1.25000) V
ADC0 = -1.11027 (-1.11842) V
ADC0 = -0.97954 (-0.98684) V
ADC0 = -0.84866 (-0.85526) V
ADC0 = -0.71790 (-0.72368) V
ADC0 = -0.58703 (-0.59211) V
ADC0 = -0.45626 (-0.46053) V
ADC0 = -0.32548 (-0.32895) V
ADC0 = -0.19472 (-0.19737) V
ADC0 = -0.06396 (-0.06579) V
ADC0 = 0.06683 (0.06579) V
ADC0 = 0.19757 (0.19737) V
ADC0 = 0.32840 (0.32895) V
ADC0 = 0.45915 (0.46053) V
ADC0 = 0.58994 (0.59211) V
ADC0 = 0.71992 (0.72368) V
ADC0 = 0.82826 (0.85526) V
ADC0 = 0.85060 (0.98684) V
ADC0 = 0.83543 (1.11842) V
ADC0 = -1.24074 (1.25000) V
```

The number is parenthesis is the DAC output voltage minus 1.25 V.
We see that the reading is wrong for a DAC voltage above 2.1V (0.85 V + 1.25 V).

The ADC (AD7124-8) is supplied between -1.8 V and +1.8 V.
The datasheet specifies a maximum input voltage of 0.3 V beyond the rails, that is 2.1 V.
There is a 1 kOhm resistor in series with the ADC input, so when the input protection diode starts
conducting current a voltage drop occurs in the 1 kOhm and the measure voltage starts to drop.

To summarize (this is missing in the user guide; I will add it):

- Common mode voltage should be between -1.8 V and +1.8V (so the 1.25V you are using are fine).
- Voltage on any ADC input pin should also be kept between -1.8 V and +1.8V.
  
Now, looking at your data, this doesn't explain everything.
In your case maximum DAC voltage is 1.84 V and I don't observe a drift on the 1.25 V. 
This may be because your using simultaneously all the channels.
Also which kind of capacitor do you use for the 100 uF ?

grahamnaylormrc, 2019-02-01:
Thanks Thomas - that make sense. It is  a lot more reliable when limiting the DAC to 2V in this particular arrangement (ie the DACs are OK up to 2.5V, but the ADC won't be OK above the power rails). I had originally looked at the spec sheet for the ADC where it says inputs should be kept to +-vref/gain and I took the gain to be 1. So the take home message is that the voltage on the inputs should both be in the range +/-1.8V as even with the allowable 0.3V beyond the rails, there was occasionally erratic behaviour.
Now I understand the limits, this should work great for our application.
thanks very much.
Graham

closed 2019-02-01 by grahamnaylormrc