Koheron/koheron-sdk #496: recommended paths for instrument config.yml

okz, 2019-05-13:
Hi, 

I am working on our own instruments projects, and my thinking it to keep our instrument project, and the koheron-sdk seperate, in their own paths. i.e: 
```
> $koheron-sdk/ 
> $my-instrument/ 
```
After using relative paths in our config.yml file, the instrument can be built with: 
> koheron-sdk$>make CONFIG="../my-instrument/config.yml" 

Seperating the folders, and using the relative paths allows us to maintain the sdk and our projects seperately. So all is well. 

However,  the make script inside the sdk ends up creating the tmp folder as: 
> $tmp/../my-instrument

and since we have to start make inside the koheron-sdk path, the instrument folder ends up created inside the sdk folder, and an empty tmp folder. 

The examples I have seen, always had the instrument files, inside the sdk, which is ok for sdk examples, but probably not good for inhouse projects. 

Is there a way around this? Or should the make file be updated to create the tmp folder correctly? 

Kind regards,
Oz

jeanminet, 2019-05-13:
Hi Oz,

You can use a top Makefile that will include the Makefile of the sdk:

```make
SDK_URL := git@github.com:Koheron/koheron-sdk
SDK_TAG := master
SDK_PATH := build
CONFIG := my-instrument/config.yml

.PHONY: install
install:
	git clone $(SDK_URL) $(SDK_PATH) || true
	git -C $(SDK_PATH) checkout $(SDK_TAG)
	git -C $(SDK_PATH) pull origin $(SDK_TAG)

.PHONY: update
update:
	git -C $(SDK_PATH) checkout $(SDK_TAG)
	git -C $(SDK_PATH) pull origin $(SDK_TAG)

-include $(SDK_PATH)/Makefile
```

You can then run `$ make install` and `$ make setup` to install the sdk in a separate `/build` folder.

Let me know if that works for you,
Jean

okz, 2019-05-14:
Hi Jean, 

This is quite neat..
Thank you. 

Oz

closed 2019-05-14 by okz