Koheron/koheron-sdk #504: Registers for Koheron alpha250

AnthoF62, 2019-10-25:
Hi,

I am using the Alpha250 and I would like to use the Precision DACs, ADCs and the clock generator. I saw that they were all configured by the SPI bus. I would like to talk to them via SPI through Python but unfortunately I don't see the registers. Where could I find them please ?

Thanks in advance,
Anthony Fortin
2M Engineering

rsarwar87, 2019-10-26:
Hi

Check boards/alpha250/drivers/. drivers are all there.

You have to do is expose those functions with appropriate python scripts.
refer to drivers_table.hpp in tmp/examples/alpha250/adc-dac-dma/server

Best Regards
Rashed

AnthoF62, 2019-11-06:
Hi,

Thanks for the answer, 
I unfortunately don't find the drivers_table.hpp file. 
So if I understand well, I need to "translate" these 3 drivers files (plus the ones they are using) into Python, right ?

Thanks in advance,
Regards
Anthony

tvanderbruggen, 2019-11-06:
Hi,

If I understand properly, you want to use the Precision DACs, ADCs and the clock generator from a python script running on the Zynq.

I that case the simple way is probably to build and run an instrument that includes the required drivers. Make sure to include to required drivers in your instrument `config.yml`:
```yaml
drivers:
- ...
- boards/alpha250/drivers/precision-adc.hpp
- boards/alpha250/drivers/precision-dac.hpp
- boards/alpha250/drivers/clock-generator.hpp
- boards/alpha250/drivers/spi-config.hpp"
```

Then you can call the drivers by accessing the server localy. Your python script running on the Zynq  would be something like that (assuming your instrument is already running):

```py
from koheron import command, KoheronClient

class MyInstrument(object):
    def __init__(self, client):
        self.client = client

    @command(classname='ClockGenerator')
    def set_sampling_frequency(self, val):
        pass

# Connect via TCP to localhost (IP 127.0.0.1):
client = KoheronClient( '127.0.0.1')
# Alternatively you can use an Unix socket (less overhead than TCP for local connections):
# client = KoheronClient(unixsock='/var/run/koheron-server.sock')

driver = MyInstrument(client)

# Call the ClockGenerator driver and set the sampling frequency
driver.set_sampling_frequency(0)
```

Does this answer your question ?

AnthoF62, 2019-11-08:
Hi,

Thanks for the answer,

Yes it is indeed what I would like to do, as I would like to set a sampling frequency for two outputs : CLKout0_DIV of 250 MHz and then another output clock of around 100 kHz, to receive the datas and store them with the precision ADC's at a rate of 250 MHz  and use the precision DAC's with a frequency of 100 kHz, all through Python.

Is it possible to use an already created class ? Like FFT for example which has already the required drivers (spi-config.hpp, clock-generator.hpp, precision-adc.hpp and precision-dac.hpp) ?

Also how do I know which commands I need and how do I write them, please ?

Thanks for the help !

tvanderbruggen, 2019-11-08:
Yes can use an already created instrument from the [examples](https://github.com/Koheron/koheron-sdk/tree/master/examples/alpha250). They all have the required drivers. Try to select the instrument which is the closest from your target application.

From what I understand you need an ADC with sampling rate of 250 MHz. You will have to use one of the 2 RF ADC for that. The precision ADC has a maximum sampling rate of 19.2 ksps. All the example instruments configure by default the clock generator for sampling rates of 250 Msps.

Concerning the precision DAC, you can operate it at 100 ksps (note that the settling time is 5 µs). The precision DAC clock is not set by the clock generator but directly by the FPGA (LDAC pin). If you want to stream a waveform at this rate on the precision DAC you will have to implement the required logic in front of the [Precision DAC SPI core](https://github.com/Koheron/koheron-sdk/tree/master/boards/alpha250/cores/precision_dac_v1_0).

Mohammed-678, 2023-06-27:
Hi I have a problem sending my bitstream file to the ALPHA250 card I configured  ethernet connection between a host and a Zynq board as precised on koheron website but I don't know what is the next step