Koheron/koheron-sdk #508: __init.py__ and ubuntu18 image

rsarwar87, 2020-02-05:
Hi, I have managed to get support for ultrascale boards within the framework which is able to compile the bit.bin as well as pl.dtbo which updates the clock of the FPGA. its based off ubuntu 18; however, the /usr/local/app/__init__.py seems to run into compatibility issues with ubuntu 18.04.

any idea how to fix it?

````thunder lock: disabled (you can enable it with --thunder-lock)
uwsgi socket 0 bound to UNIX address /var/run/uwsgi/uwsgi.sock fd 6
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
Python version: 3.6.9 (default, Nov  7 2019, 10:44:02)  [GCC 8.3.0]
Python main interpreter initialized at 0x5591ed1f90
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 145808 bytes (142 KB) for 1 cores
*** Operational MODE: single process ***
zip_filename /usr/local/instruments/default.ZIP
target_filename version <class 'str'>
zip_files type <class 'str'> len 1596
Traceback (most recent call last):
  File "/usr/local/api/wsgi.py", line 1, in <module>
    from app import app as application
  File "./app/__init__.py", line 115, in <module>
    app = KoheronApp(__name__)
  File "./app/__init__.py", line 56, in __init__
    self.init_instruments(KoheronApp.instruments_dirname)
  File "./app/__init__.py", line 100, in init_instruments
    self.run_instrument(instrument_filename, KoheronApp.live_instrument_dirname, instrument)
  File "./app/__init__.py", line 108, in run_instrument
    print('Installing instrument ' + name)
TypeError: must be str, not NoneType
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
*** uWSGI is running in multiple interpreter mode ***
```

tvanderbruggen, 2020-02-05:
It might be because your are calling the script with python 3 and it was develop for python 2. There is some differences for strings in particular ...

rsarwar87, 2020-02-06:
Thanks for that, i moved it to python2 and how it works.

Just realized that when loading the bitstream manually, everything seems to work perfectly. 

May I ask how you were planning to update the clocks on fpga or any kernel above 4.9? it seems to behave inconsistently from one device to another. so far i only had success with one ultra-scale device using 2019.1

tvanderbruggen, 2020-02-06:
Yes we have some improvements planned but unfortunately not much time to work on it recently.

Main plan is to launch the FPGA bitstream and program the clocks from the C++ app.
It will remove lots of boilerplate (ex. using systemd to sequence the startup), get a more deterministic startup and improve error messages when something goes wrong.

I added the clock configuration into the C++ (https://github.com/Koheron/koheron-sdk/blob/v0.20/server/context/zynq_fclk.hpp) but still have to do the bitstream startup.
All of this is done with ubuntu 16.04 so the command will have to be updated to accommodate the new drivers, but from what I saw you already did this part of the work.

So I would like to finish these architecture changes with ubuntu 16.04 and then migrate to 18.04 (basically merge your work).

rsarwar87, 2020-02-06:
Yes I say that, however, that works for zynq and not zynqmp from what I gather. 

From what I know, the only way, correct me if I am wrong, to change the clocks on devices using fpga_manager or ultrascale from linux is to use overlays.

```  fragment@1 {                                                                                                                     
 10     target = <&amba>;                                                                                                            
  9     overlay1: __overlay__ {                                                                                                       
  8       afi0: afi0 {                                                                                                               
  7         compatible = "xlnx,afi-fpga";                                                                                            
  6         config-afi = < 0 0>, <1 0>, <2 0>, <3 0>, <4 0>, <5 0>, <6 0>, <7 0>, <8 0>, <9 0>, <10 0>, <11 0>, <12 0>, <13 0>
  5       };                                                                                                                           
  4       clocking0: clocking0 {                                                                                                       
  3         #clock-cells = <0>;                                                                                                        
  2         assigned-clock-rates = <99999001>;                                                                                         
  1         assigned-clocks = <&zynqmp_clk 71>;                                                                                        
31          clock-output-names = "fabric_clk";                                                                                         
  1         clocks = <&zynqmp_clk fabric_clk [A]                                                                                       
  2         compatible = "xlnx,fclk";                                                                                                  
  3       };                                                                                                                           
  4       clocking1: clocking1 {                                                                                                 
  5         #clock-cells = <0>;                                                                                                   
  6         assigned-clock-rates = <49999500>;                                                                                       
  7         assigned-clocks = <&zynqmp_clk 72>;                                                                                        
  8         clock-output-names = "fabric_clk";                                                                                      
  9         clocks = <&zynqmp_clk 72>;                                                                                             
 10         compatible = "xlnx,fclk";                                                                                                  
 11       };                                                                                                                 
 12     };                                                                                                                          
 13   };    ```

if that is indeed the case, it would be nice to have this included as well. I was planning to do it.
https://github.com/Xilinx/meta-xilinx-tools/blob/master/recipes-bsp/fpga-manager-script/files/fpgautil.c
for now, I added it to the install_instrument.sh script

tvanderbruggen, 2020-02-06:
We haven't experienced with the ultrascale yet.

Your install_instrument.sh file is very neat. I will integrate what you did.

Thanks.

rsarwar87, 2020-02-06:
Yes. I think it would be a good adition, and allow us to move forward to latest kernels. I see you have pushed a lot today, including a fpgamanager. I will see if I can add support for overlays. 

re: ultrascale; i have been trying to port ultrascale devices for a while now:

https://github.com/rsarwar87/koheron-sdk/tree/ultrascale

Most of the HDL related tcls are working properly. as well as creation of overlays. I have not ventured too much into building BOOT.BINs, but it should not be possible.

This branch also contains new board supports, including enclustra mar+ zx3, pe1+Ux1 and pe1+ZX1; and trenz tebf808+te803.

tvanderbruggen, 2020-02-07:
We have this branch https://github.com/Koheron/koheron-sdk/pull/486 which will include support for latest kernels and for ultrascale. The FPGA manager and the overlay needs to be added on this branch. It  would be useful to compare your change with this branch. 

I need to finish merging some stuff on master, that will clean up the diff...

rsarwar87, 2020-02-07:
Hi, I will have a look, i did not see this branch before. thanks.

Just a heads up, xilinx has changed some of the implementations of creating overlays moving from 2018.2 to 2018.3. It is more stable but requires a slightly different process.

I tried cloning your fork of the xilinx kernel, and rebase it with Xilinx 2019.2 release, but there were too many conflicts. I'll give another go later.