Koheron/koheron-sdk #509: Python problem, maybe?

gitul, 2020-02-11:
Hi,
I do not have a lot of experience with FPGAs, and just started to work on the alpha250 board. 

For now, all I want is to make the default FFT instrument work. As far as I know, all the files for this instrument are already loaded on the alpha250 when you purchased the board. So, I have not copied files to the alpha250.

What I have done is try to run some of the Python FFT examples. I installed the koheron Python module, which went fine. My Python version is 3.6.9. The connection between my Ubuntu PC and the alpha250 seems to be working fine.

My problem happens when I try to read a PSD. I get the error shown below.

Any help would be welcome
Thank you
Raul

```
In [81]:
In [81]: driver.read_psd_raw()                                                                                       
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-81-63b695fc185e> in <module>
----> 1 driver.read_psd_raw()

~/.local/lib/python3.6/site-packages/koheron/koheron.py in wrapper(self, *args)
     74             self.client.last_device_called = device_name
     75             self.client.last_cmd_called = cmd_name
---> 76             return func(self, *args)
     77         return wrapper
     78     return real_command

~/work/koheron/koheron-sdk-master/examples/alpha250/fft/python/fft.py in read_psd_raw(self)
     46     @command()
     47     def read_psd_raw(self):
---> 48         return self.client.recv_array(self.n_pts/2, dtype='float32')
     49 
     50     # DDS

~/.local/lib/python3.6/site-packages/koheron/koheron.py in recv_array(self, shape, dtype, check_type)
    426         self.recv(fmt='')
    427         buff = self.recv_all(dtype.itemsize * arr_len)
--> 428         return np.frombuffer(buff, dtype=dtype.newbyteorder('<')).reshape(shape)
    429 
    430     def recv_tuple(self, fmt, check_type=True):

TypeError: 'float' object cannot be interpreted as an integer

In [82]:
```

jeanminet, 2020-02-11:
Hi Raul,
Can you try running the script with Python 2 (`$ python2 fft.py`) to check if your problem is related to your Python version?

gitul, 2020-02-11:
Hi Jean,
I just tried with Python 2.X and it worked. So, it is a Python 3.X issue.

Is Koheron-python meant to be Python 3 compatible? Probably not? If not, are there any plans to make it compatible?

Thank you very much.
Raul

ps: I have more questions but will create another thread.