Koheron/koheron-sdk #510: PSD average on the Alpha250

gitul, 2020-02-11:
Hi Jean,

In my application (with the alpha250), I need to average in time many PSDs as efficiently as possible (highest cadence possible).

Does the Python software provided by Koheron to obtain a PSD have the functionality to average many consecutive PSDs  on the FPGA ?  In other words, something like:

```
# Number of consecutive PSDs to average on the FPGA
Npsd = 10000

# Produce averaged PSD
average_psd = driver.read_psd_raw( Npsd )
```

Thanks in advance
Raul

jeanminet, 2020-02-12:
Hi Raul,
This functionality is not implemented in the programmable logic and thus not accessible from Python.

gitul, 2020-02-12:
OK, thanks for the response.
Raul

closed 2020-02-12 by gitul