Koheron/koheron-sdk #511: Number of points for FFT computation on Alpha250

gitul, 2020-02-12:
Hi Jean,

For the alpha250, I know it is possible to determine the number of points used (in the time domain) for the FFT computation as follows:

```
n_pts = driver.n_pts    # in time
```
However, is there a way to 'set' the number of points used? . For instance, instead of using 2^13 points (the default), I would like to use 2^15.

Thank you very much
Raul

jeanminet, 2020-02-12:
Hi Raul,
The number of points used is hard wired in the FPGA.
You need to recompile a FPGA bitstream to change it.
Unfortunately, 2^13 points is the maximum FFT size that can fit on the programmable logic.

gitul, 2020-02-12:
OK, thanks Jean. We can still work with 2^13 points.
Raul

closed 2020-02-12 by gitul