Koheron/koheron-sdk #512: Alpha250 Fan not turning on

gitul, 2020-02-12:
Hi Jean,

When I use the Alpha250, the fan never turns on. After a few minutes the heatsink becomes dangerously hot to the touch. I know that the board should be able to withstand high temperatures, but it seems suspicious to me not seeing the fan turning on.

I couldn't find detailed information on the website/user manual about this.

Your guidance would be appreciated.
Raul

jeanminet, 2020-02-12:
Hi Raul,
This is not a software issue, the fan should be always on.
Can you check the fan electrical connection?
It may have disconnect during shipping.

gitul, 2020-02-12:
Hi Jean,
The connection was mechanically very tight. That is why I was hesitant to disconnect it before.

But now I did it, and after reconnection the fan started working.

Thank you.
Raul

closed 2020-02-12 by gitul