Koheron/koheron-sdk #513: instrument updates fail on newly built images

okz, 2020-03-12:
Hi, 
 
Having some problems on a new sdk build image, and wanted to check if it's due to an updated external dependency. 

Has a new image build been tested lately for updating an instrument ? 

Had problems on our own projects, but some of them can be replicated using the sdk examples: 

Have a previously built alpha250 image, running this on the board, and building the fft example using:
```
    make CONFIG=./examples/alpha250/fft/config.yml HOST=192.168.86.168 run
```
The instrument builds and is updated on the board. 

Having build a new clean image, it hasn't been working however: 

    make clean
    make CONFIG=./examples/alpha250/fft/config.yml 
    make CONFIG=./examples/alpha250/fft/config.yml os
    sudo CONFIG=./examples/alpha250/fft/config.yml image
    
at this stage, using the image on the board, the example runs correctly.  However have never been able to install a new instrument on the new image:

    make CONFIG=./examples/alpha250/fft/config.yml HOST=192.168.86.168 run

I get: 

```
*   Trying 192.168.86.168...
* TCP_NODELAY set
* Connected to 192.168.86.168 (192.168.86.168) port 80 (#0)
> POST /api/instruments/upload HTTP/1.1
> Host: 192.168.86.168
> User-Agent: curl/7.58.0
> Accept: */*
> Content-Length: 710419
> Content-Type: multipart/form-data; boundary=------------------------be57d2f4f885d2b1
> Expect: 100-continue
> 
< HTTP/1.1 100 Continue
< HTTP/1.1 500 Internal Server Error
< Server: nginx/1.10.3 (Ubuntu)
< Date: Thu, 12 Mar 2020 23:24:25 GMT
< Content-Type: text/plain
< Transfer-Encoding: chunked
< Connection: keep-alive
* HTTP error before end of send, stop sending
< 
* transfer closed with outstanding read data remaining
* stopped the pause stream!
* Closing connection 0
curl: (18) transfer closed with outstanding read data remaining
Makefile:78: recipe for target 'run' failed
make: *** [run] Error 18
```

looking at the error log on the board as well:

==> /var/log/nginx/error.log <==
```
2020/03/12 22:48:10 [error] 1492#1492: *4 readv() failed (104: Connection reset by peer) while reading upstream, client: 192.168.86.169, server: localhost, request: "POST /api/instruments/upload HTTP/1.1", upstream: "uwsgi://unix:/var/run/uwsgi/uwsgi.sock:", host: "192.168.86.168"
2020/03/12 22:57:42 [error] 1492#1492: *7 readv() failed (104: Connection reset by peer) while reading upstream, client: 192.168.86.169, server: localhost, request: "POST /api/instruments/upload HTTP/1.1", upstream: "uwsgi://unix:/var/run/uwsgi/uwsgi.sock:", host: "192.168.86.168"
2020/03/12 23:24:25 [error] 1492#1492: *9 readv() failed (104: Connection reset by peer) while reading upstream, client: 192.168.86.169, server: localhost, request: "POST /api/instruments/upload HTTP/1.1", upstream: "uwsgi://unix:/var/run/uwsgi/uwsgi.sock:", host: "192.168.86.168"
```

Kind regards,
Oz

tvanderbruggen, 2020-03-13:
Hi Oz,

Thanks for reporting the problem.

Did you build the image from the master branch ? We've made some changes on the OS side that require to rebuild the instruments.

Can you send me the result of:

```
journalctl -u koheron-server -n 300
```

on the board ?

Thank you.

Regards,

Thomas

okz, 2020-03-13:
Hi Thomas, thanks for the quick response.

Yes it was built from the master branch.  But the example instrument's were built using the same branch. 

Is there a recommended branch tag I should be using ?  

I am building things to get  journal log's again. Should have backed up test builds...

Kind regards,
Oz

okz, 2020-03-13:
Hi Thomas, 

Copied the koheron-server log below.  This is done using a new build from the master branch, using: 

```
make clean
make CONFIG=./examples/alpha250/fft/config.yml 
make CONFIG=./examples/alpha250/fft/config.yml os
sudo CONFIG=./examples/alpha250/fft/config.yml image
```

-- Logs begin at Thu 2016-02-11 16:28:01 UTC, end at Fri 2020-03-13 12:39:56 UTC. --
Feb 11 16:28:04 koheron systemd[1]: Starting Koheron TCP/Websocket server...
Feb 11 16:28:04 koheron serverd[1427]: FpgaManager: Loading bitstream /tmp/live-instrument/fft.bit...
Feb 11 16:28:04 koheron serverd[1427]: FpgaManager:Bitstream successfully loaded
Feb 11 16:28:04 koheron serverd[1427]: ZynqFclk: Clock fclk0 set to 200000000 Hz
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [2] Common...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [7] ClockGenerator...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [3] Eeprom...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [12] SpiConfig...
Feb 11 16:28:04 koheron koheron-server[1427]: Clock generator - Not initialized
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [4] GpioExpander...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [8] Ltc2157...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [9] Ad9747...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [11] PrecisionDac...
Feb 11 16:28:04 koheron koheron-server[1427]: Driver Manager: Starting driver [10] PrecisionAdc...
Feb 11 16:28:04 koheron koheron-server[1427]: Common - Initializing ...
Feb 11 16:28:04 koheron koheron-server[1427]: Clock generator - Setting default configuration ...
Feb 11 16:28:04 koheron koheron-server[1427]: Clock generator - TCXO calibration is 158
Feb 11 16:28:04 koheron koheron-server[1427]: Clock generator - Ref: TCXO, VCO: 2500.000000 MHz, ADC: 250.000000 MHz, DAC: 250.000000 MHz
Feb 11 16:28:04 koheron koheron-server[1427]: Koheron server ready
Feb 11 16:28:04 koheron systemd[1]: Started Koheron TCP/Websocket server.
Feb 11 16:28:05 koheron koheron-server[1427]: Interface eth0 found: 192.168.86.168
Feb 11 16:28:06 koheron koheron-server[1427]: TCPSocket: Connection closed by client
Mar 13 12:35:01 koheron koheron-server[1427]: Driver Manager: Starting driver [13] FFT...
Mar 13 12:35:01 koheron koheron-server[1427]: Driver Manager: Starting driver [5] TemperatureSensor...
Mar 13 12:35:01 koheron koheron-server[1427]: Driver Manager: Starting driver [6] PowerMonitor...
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close
Mar 13 12:35:14 koheron koheron-server[1427]: WebSocket: Connection close

tvanderbruggen, 2020-03-13:
Thanks for the data. 

I'm building a fresh image from master and try to reproduce the issue.

I keep you updated.

Regards,

Thomas

tvanderbruggen, 2020-03-13:
I managed to reproduce the problem.

UWSGI crashes with

```
Feb 11 16:28:07 koheron uwsgi[1450]: Traceback (most recent call last):
Feb 11 16:28:07 koheron uwsgi[1450]:   File "/usr/local/api/wsgi.py", line 1, in <module>
Feb 11 16:28:07 koheron uwsgi[1450]:     from app import app as application
Feb 11 16:28:07 koheron uwsgi[1450]:   File "./app/__init__.py", line 5, in <module>
Feb 11 16:28:07 koheron uwsgi[1450]:     from werkzeug import secure_filename
Feb 11 16:28:07 koheron uwsgi[1450]: ImportError: cannot import name secure_filename
```

This doesn't seem to come from a change we have made but from a change in the werkzeug package.

[Here](https://github.com/GoogleCloudPlatform/getting-started-python/issues/256) they recommend to replace 

```
from werkzeug import secure_filename
```

by

```
from werkzeug.utils import secure_filename
```

I will test this fix and publish it.

tvanderbruggen, 2020-03-13:
I've tested it (8552d19) and it seems to work now.

Thanks for pointing out the problem.

Regards,

Thomas

okz, 2020-03-13:
Thanks Thomas, can now upload our own instruments as well. 

It also explains why even old sdk release builds were now having problems. 

Kind regards
Oz

closed 2020-03-16 by tvanderbruggen