Koheron/koheron-sdk #518: MMCM

grahamnaylorccfe, 2020-06-01:
Is there any documentation anywhere on how to set the mmcm register? Presumably this sets the phase and source of the ADC sampling?
thanks
Graham

tvanderbruggen, 2020-06-01:
Hi Graham,

You can use the adc-dac-bram design and the related [script](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-bram/test_phase_shift.py) to determine the MMCM phase-shift for a given sampling frequency.

grahamnaylorccfe, 2020-06-04:
Thanks for this Thomas. Is the procedure to initially set the value to 0 for mmcm in the clock configuration being tested and then from the plot find the value where the noise is lowest and flat. Then use that value as the mmcm value in the clock configuration in clock-generator.hpp?
cheers Graham

closed 2020-09-18 by grahamnaylorccfe