Koheron/koheron-sdk #526: Clock Configuration

grahamnaylorccfe, 2020-09-18:
I have been adding clock configurations to a copy of your clock-generator.hpp driver from version 0.20 see below.
Clock configurations 2-9 work fine but clock configurations 0 and 1 (which are simply copies of configs 10 and 11) show a very strange behaviour as if the ADC sampling becomes irregular. I always keep the ADC, DAC and FPGA clocks at the same frequency in each configuration and in the range 170-250MHz. The clock configs with strange ADC behaviour tend to have lower sample rates. I do two tests.
i) Firstly I output a sinewave generated to the DAC to test its operation all configs operate correctly except configs 0,1 and 10 that outputs a highly distorted wave (as if the DAC is not sampling uniformly). configs 0 and 1 should be the same as 10 and 11, however they behave differently as the clock output on the SMA (CLK0) is at 10MHz instead of the expected frequency in the 150-200MHz range.
ii) I input a test sinewave at around 160MHz into channel 0 and in the FPGA direct it to the DAC and examine on a scope. I expect an aliased frequency of a few tens of MHz and a sinewave. This works as expected for configs 2-9, but 0,1 and 10,11 give very ragged (but repetetive) waveforms at unexpected frequencies.

Is there a potential issue of having 12 clock configurations?

Excerpt of clock-generator driver:

namespace clock_cfg {

    // Input clock selection
    constexpr uint32_t EXT_CLOCK = 0;
    constexpr uint32_t FPGA_CLOCK = 1;
    constexpr uint32_t TCXO_CLOCK = 2;
    constexpr uint32_t PIN_SELECT = 3;
    constexpr uint32_t AUTO_CLOCK = 4;

    constexpr auto clkin_names = koheron::make_array(
        koheron::str_const("External"),
        koheron::str_const("FPGA"),
        koheron::str_const("TCXO"),
        koheron::str_const("Pin Select"),
        koheron::str_const("Auto")
    );

//Need to adjust num_configs if extra clock settings are added
    constexpr size_t num_configs = 12;
    constexpr size_t num_params = 10;

//PLL2_P= 5 ,PLL2_N= 58, PLL2_R=  24.0 Fre_VCO = 2416.6666666666665
// AOM freq= Freq_VCO/15 = 161.111..
    // Sampling frequency 2416.6666666666665/11=219.696969.. MHz 
   // Fvco= 100*2*PLL2_N*PLL2_P/PLL2_R    -> PLL2_P is the prescalar and can be 2-8, whereas PLL2_N can be 1-262143 and PLL2_R can be 1-4095
   // DDS on fpga will therefore need to be set to 4*Fadc/15 = ~58.58..MHz   clock config 0
	//mmcm value may need checking!

    constexpr std::array<uint32_t, num_params> Ammonia0 = {
        5,   // PLL2_P
        58,   // PLL2_N
        24,   // PLL2_R
        15, // CLKout0_DIV (CLKOUT) - this goes to the SMA
        11,  // CLKout1_DIV (ADC clock)
        11,  // CLKout2_DIV (DAC clock)
        11,  // CLKout3_DIV (FPGA clock)
        237, // CLKout4_DIV (EXP_CLK0 clock)
        237, // CLKout5_DIV (EXP_CLK1 clock)
        30   // MMCM phase shift
    };

//PLL2_P= 5 ,PLL2_N= 58, PLL2_R=  24.0 Fre_VCO = 2416.6666666666665
// AOM freq= Freq_VCO/15 = 161.111..

    // Sampling frequency 2416.6666666666665/14=172.619.. MHz 
   // Fvco= 100*2*PLL2_N*PLL2_P/PLL2_R    -> PLL2_P is the prescalar and can be 2-8, whereas PLL2_N can be 1-262143 and PLL2_R can be 1-4095
   // DDS on fpga will therefore need to be set to Fadc/15 = ~11.5..MHz   clock config 1
	//mmcm value may need checking!

    constexpr std::array<uint32_t, num_params> Ammonia0b = {
        5,   // PLL2_P
        58,   // PLL2_N
        24,   // PLL2_R
        15, // CLKout0_DIV (CLKOUT) - this goes to the SMA
        14,  // CLKout1_DIV (ADC clock)
        14,  // CLKout2_DIV (DAC clock)
        14,  // CLKout3_DIV (FPGA clock)
        237, // CLKout4_DIV (EXP_CLK0 clock)
        237, // CLKout5_DIV (EXP_CLK1 clock)
        30   // MMCM phase shift
    };
[... 168 lines cut ...]
        5,   // PLL2_P
        58,   // PLL2_N
        24,   // PLL2_R
        15, // CLKout0_DIV (CLKOUT) - this goes to the SMA
        14,  // CLKout1_DIV (ADC clock)
        14,  // CLKout2_DIV (DAC clock)
        14,  // CLKout3_DIV (FPGA clock)
        237, // CLKout4_DIV (EXP_CLK0 clock)
        237, // CLKout5_DIV (EXP_CLK1 clock)
        30   // MMCM phase shift
    };

    constexpr std::array<std::array<uint32_t, num_params>, num_configs> configs = {Ammonia0, Ammonia0b, CH4_Clads, CH4_LDS100, CO2, Richard, CH4_LDS100_offset, CH4_LDS100_offset2, CO2_LDS100, Isotope, Ammonia1, Ammonia2 };
}

class ClockGenerator
{
  public:
    ClockGenerator(Context& ctx_)
    : ctx(ctx_)
    , ctl(ctx.mm.get<mem::control>())
    , i2c(ctx.i2c.get("i2c-0"))
    , eeprom(ctx.get<Eeprom>())
    , spi_cfg(ctx.get<SpiConfig>())
    {...........................

tvanderbruggen, 2020-09-18:
Hi Graham,

I see that the troublesome configurations are the ones with a `//mmcm value may need checking!` comment.
Did you run the [test_phase_shift.py](https://github.com/Koheron/koheron-sdk/blob/master/examples/alpha250/adc-dac-bram/test_phase_shift.py) for those configs ?

Also the driver logs the programmed frequencies for the reference, the VCO, the ADC and the DAC.
Do they match the values you expect ?

The clock output onthe SMA is an LVCMOS output. Its maximum frequency is 250 MHz (according to LMK04906 datasheet, p.11). so 150-200MHz should be ok.

I don't see why having 12 configurations would be an issue.

grahamnaylorccfe, 2020-09-18:
Hah yes. Thats the one. Should read my own comments !
Thanks alot.
Graham

closed 2020-09-18 by grahamnaylorccfe

grahamnaylorccfe, 2020-09-22:
In case anyone stumbles across this in the future and wishes to optimise the MMCM value defined in the clock-generator.hpp file, here are a few tips:
The value can be tuned using the adc-dac-bram instrument provided , but pointing to your version of the clock-generator driver in the config.yml file. Start with the mmcm value set to 0 for the clock config to be tested. Run the python utility ‘test_phase_shift.py’ (first editing the clk_ definitions at the end of the program to reflect the config to be tested and the ip address of the board on which the test is being done). There should be no signal input to the ADCs when running this test. A plot with a square wave will appear. Choose a value of the x axis where the values are closest to zero (bottom of the square wave) in the middle of a bath tub. Enter this value as the MMCM value in the clock-generate.hpp file. This clock definition can then be used in other instruments to be developed.

jeanminet, 2020-09-22:
Thanks Graham,
To illustrate, here is what you get when you set the phase shift to zero on the standard 250 MHz clock:

[image]

The optimal phase shift (the middle of the bath tub) is here about 40.

Note: The phase shift in the reference design is set to 56:
https://github.com/Koheron/koheron-sdk/blob/master/boards/alpha250/drivers/clock-generator.hpp#L58