Koheron/koheron-sdk #528: Ethernet eth0 DMA bus error: HRESP not OK

afjmirico, 2020-10-01:
Alpha 250 SDK 0.20
This error was encountered previously and was fixed by amending the line in the board.mk file
LINUX_TAG := koheron-v$(VIVADO_VERSION)-kernel-module-fix-dma

Having switched to the latest SDK, we have now started seeing this issue again even though the board.mk file still has the modification in it. Our system sporadically copies large amounts of data from the Alpha250 onto other processing modules across ethernet and we see this error once or twice a day when in operation. To recover requires a reboot or sometimes a power cycle.

Recreating the issue in a controlled way is challenging. In tests it has taken between 1 and 3 hours whilst continually copying data from the Alpha 250 to observe the problem. We continue to test.

tvanderbruggen, 2020-10-02:
[Here](https://forums.xilinx.com/t5/Embedded-Linux/eth0-DMA-bus-error-HRESP-not-OK-with-2019-2/td-p/1118859) they suggest to increase the CMA pool.

Some [notes](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841683/Linux+Reserved+Memory#LinuxReservedMemory-ReservedmemoryforCMA) from Xilinx

afjmirico, 2020-10-07:
We have been building using directory tree parallel to the koheron-sdk directory which includes it's own board.mk file which had the fix-dma tag included in it.
To try to force the build to use this file, we deleted the board.mk file from the koheron-sdk, but this does not work. The location of the board.mk file is always within the koheron-sdk directory. If it does not exist, the build system does not produce an error, but proceeds without any tags. That is because the variables are set in the os.mk file (lines 23-25) which also dictates the path to the board.mk file (lines 17-21).
So the problem, as reported, was because of our lack of understanding of the build system rather than anything else.
Modifying the board.mk file in koheron-sdk results in the fix being included and we no longer see the DMA errors.

closed 2020-10-07 by afjmirico

jeanminet, 2020-10-08:
Fixed with #529