Koheron/koheron-sdk #536: How to apply the phase noise analysis SDK to the Alpha250

qiuquanyan, 2021-03-26:
Hello,

I'd like to ask a question. The FFT and precision ADC/DAC could be used by visiting the network. But I don't know how to use the phase noise analysis (PNA) SDK in the Alpha250. Could you please tell me an operating process for using the PNA?

Thanks very much!

jeanminet, 2021-03-26:
Hi,

You can download the .zip file at this link:
https://www.koheron.com/fpga/reference-designs/
then use this web page to add the instrument:
[image]

qiuquanyan, 2021-03-28:
Hi,

Thanks for your quick reply! Now I can use phase noise analysis normally.

qiuquanyan, 2021-03-29:
Hi, Jean.

When using phase noise analysis, we encountered two problems as follows:
1. We cannot set the carrier center frequency on the web page.
2. We don’t know how to derive the phase noise analysis data from this webpage.

Could you please help me answer these questions?
Thanks!